nasa/fprime #479: This is not even pre-modern C++

DBJDBJ, 2021-04-19:
And that it is perfectly understandable. Of course. I had time for 15 min fly over. Software is clear, simple and (I believe) resilient. All good things.

I also understand this was ready-made, running (before) on VxWorks, mature etc. I assume somebody has measured Benefits vs Risks ... 

But here is the elephant in the room: Why not C99?

Grabber, 2021-04-21:
Why not Rust?

Joshua-Anderson, 2021-04-22:
F' has to support a variety of embedded systems, so it needs to stick with embedded C++ best practices, particularly for the core framework (/fw). This means no stdlib, dynamic memory allocation, or templating.

Since embedded systems use a wide variety of compilers (and often outdated too!), F' tries to be conservative on the C++ features used in the core framework portions. That being said, I'd say any C++ compiler from the last 20 years should  support C++98, and I don't think anyone would have concerns with the utilizing C++98 features in F'.

Nothing in F' is targeting anything lower than C++98, so I'm not sure what you're F' C++ code you're seeing that appears to be targeting ancient C++.

Rust is a cool language and I look forward to it becoming a mature embedded systems language! F' is a C++ framework, so anything in written in rust will likely need to be a new framework.

Since this isn't an issue, I'm closing this, but if you have any questions feel free to ask on the F' mailing list!

DBJDBJ, 2021-04-22:
Well, that elephant in the room called "Why not C99?" seems truly invisible :) Many thanks for your reply.