nasa/fprime #502: Adding deregCommands option to CmdDispatcher

saba-ja, 2021-04-27:
| | |
|:---|:---|
|**_F´ Version_**| 2.0 |
|**_Affected Component_**|  Svc/CmdDispatcher |
---
## Feature Description

F Prime CommandDispacher currently does not have an option to deregister commands. This causes a duplicate command register segment fault crash when a client disconnects and reconnects back to the server.
To better understand the situation, let’s assume we have the following configuration:

[image]

`Ref` loads before `Backplane` and listens on port `4555`. When `Backplane` loads, it connects to `Ref` as a client, and all `Backplane` CMDs get registered to `Ref` `CmdDispacher` table.
Now the issue is when `Backplane` drops, its CMDs still stay registered in the `Ref` `CmdDispacher` table. When `Backplane` tries to reconnect back to `Ref` since all its CMDs are still registered in `Ref`, it causes FW_ASSERT due to duplicate opcode, causing `Ref` to crash with segment fault error.

## Rationale

It might be a good idea to add `deregCommands` (an opposite of `regCommands`) that could be used to deregister all the CMDs when a client gracefully goes down. This will allow the client to reconnect back to the server without causing duplicate opcode registration issue.

These are two ways that come to my mind to resolve this issue:

1. Check for OpCode before registering it. This could be done by the following to `compCmdReg_handler` shown below:

https://github.com/nasa/fprime/blob/e165c914b0d0240eba8164c52f8856ce90bee3dc/Svc/CmdDispatcher/CommandDispatcherImpl.cpp#L34-L49

Add this to line 35 above
```c++
// search for opcode in dispatch table
        for (U32 entry = 0; entry < FW_NUM_ARRAY_ELEMENTS(this->m_entryTable); entry++) {
            if ((this->m_entryTable[entry].used) and (opCode == this->m_entryTable[entry].opcode)) {
                return;
            }
        }

```

2. Add `deregCommands()` to Svc/CmdDispatcher. This could be achieved by the following steps:

a) 
In 
https://github.com/nasa/fprime/blob/e165c914b0d0240eba8164c52f8856ce90bee3dc/Svc/CmdDispatcher/CommandDispatcherComponentAi.xml#L17-L21

Add `compCmdDereg` port and `OpCodeDeregistered` event:

```xml
        <port name="compCmdDereg" data_type="Fw::CmdReg" kind="guarded_input" max_number = "$CmdDispatcherComponentCommandPorts">
            <comment>
            Command Deregistration Port. max_number should match dispatch port.
            </comment>
        </port>

...

        <event id="2" name="OpCodeDeregistered" severity="DIAGNOSTIC" format_string = "Opcode 0x%04X deregistered to port %d slot %d" >
            <comment>
            Op code deregistered event
            </comment>
            <args>
                <arg name="Opcode" type="U32">
                    <comment>The opcode to deregister</comment>
                </arg>          
                <arg name="port" type="I32">
                    <comment>The deregister port</comment>
                </arg>          
                <arg name="slot" type="I32">
                    <comment>The dispatch slot it was freed</comment>
                </arg>          
            </args>
        </event>
```

b)
Update [CommandDispatcherImpl.cpp](https://github.com/nasa/fprime/blob/devel/Svc/CmdDispatcher/CommandDispatcherImpl.cpp) to handle the requests.

c)
Update [Component Autocoder](https://github.com/nasa/fprime/blob/devel/Autocoders/Python/src/fprime_ac/generators/templates/component/cpp.tmpl) template to autogenerate `deregCommnds` for each component.

Joshua-Anderson, 2021-04-27:
Have you considered making command registration idempotent, so if you try and registering a command<->port mapping that already exists the command registration no-ops instead of erroring?

saba-ja, 2021-04-27:
I do not see an easy way to make `component.regCommands()` idempotent from the user perspective.
When `Backplane` loads it has to register its commands in its commandDispatch. The commands will also get passed to, in this example `Ref`, commandDispatcher which causes it to ASSERT for duplicate opCodes. 

We have to either go trough the pipeline and add some logic right before the commands are passed to the `Ref` from `BP` or make some changes in the fprime/Svc/CmdDispatcher/CommandDispatcherImpl.cpp:compCmdReg_handler to not ASSERT on duplicate opCodes as shown in the initial comment.

LeStarch, 2021-04-27:
What happens in the event of a non-graceful "exit" of a registered component?  In this case deregCommand would not be called?

In a system where you want the ability to update which component handles registered commands, is there a need to assert at all?  If not, I'd suggest making this a configuration property of the CmdDispatcher.  If `allowReregistration` is set to true, the assert is bypassed and the new registrant overrides the previous registrant.

commit 3b261353 (2021-05-18): saba-ja: fix #502
commit e240963d (2021-05-18): Merge pull request #504 from saba-ja/fix_502

Fix #502: Add option to allow reregister of opCodes
closed 2021-05-18 by LeStarch