nasa/fprime #509: Svc/BufferLogger tests

lewisjaggi, 2021-05-04:
| | |
|:---|:---|
|**Last**| |
|**BufferLogger**|   |
---
## Tests in Svc/Bufferlogger don't pass (specifically (TestErrors, LogFileValidation))

There are errors in the BufferLogger Errors.cpp file. Maybe access permissions on buf.

[image]

## How to Reproduce

1. fprime-util generate --ut
2. cd Svc/BufferLogger/
3. fprime-util check

## Expected Behavior

The tests must pass

Joshua-Anderson, 2021-05-04:
What platform are you running the tests on (OS, compiler)?
What version of fprime are you using (devel branch, 1.5.3)?

These tests are passing in CI and I just successfully re-ran them on Mac OS.
My suspension this is a build environment issue

jweadick, 2021-06-14:
I have this same issue, I'm running on WSL 1 and also using fprime-util check with fprime version 1.5.3. Reproduction of the error on WSL 1 is the same as stated above.

Joshua-Anderson, 2021-06-14:
Oh, it's a WSL 1 issue that makes sense. If I remember correctly F' doesn't work correctly on NTFS filesystems - either the case insensitivity or special characters cause problems. If you move F' from the windows partition to the WSL linux filesystem you may have batter luck.

jweadick, 2021-06-15:
I switched over to WSL 2 and switched to the Linux filesystem with no luck thus far just in case you were curious as to the outcome!

LeStarch, 2021-06-15:
It looks like (from briefly at the above UT output) that the problem is WSL is using permissions by default different from the expectation.  This is likely an issue because file creation permissions is usually configured at the user level...we should probably make that check less restrictive.

LeStarch, 2021-12-21:
I tested on WSL last night with a fresh clone without issue.  I think this has been fixed, but if it still errors open the ticket back up.

Blabla51, 2022-04-21:
Hi everyone,

I am currently facing the same issue, using the ubuntu:20.04 image on Docker (Windows)

I tried using Fprime 2.1.0 and Fprime 3.0.1 on fresh Ubuntu docker images, but both are raising the same error.

Here is the log for fprime-util check command in Fprime 3.0.1:

```
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 1
    Start 1: Svc_BufferLogger_ut_exe
1: Test command: /home/fprime/build-fprime-automatic-native-ut/bin/Linux/Svc_BufferLogger_ut_exe
1: Test timeout computed to be: 10000000
1: [==========] Running 9 tests from 4 test suites.
1: [----------] Global test environment set-up.
1: [----------] 1 test from Test
1: [ RUN      ] Test.LogNoInit
1: [       OK ] Test.LogNoInit (7 ms)
1: [----------] 1 test from Test (7 ms total)
1:
1: [----------] 3 tests from TestErrors
1: [ RUN      ] TestErrors.LogFileOpen
1: [       OK ] TestErrors.LogFileOpen (13 ms)
1: [ RUN      ] TestErrors.LogFileWrite
1: [       OK ] TestErrors.LogFileWrite (5 ms)
1: [ RUN      ] TestErrors.LogFileValidation
1: /home/fprime/build-fprime-automatic-native-ut/F-Prime/Svc/BufferLogger/GTestBase.cpp:196: Failure
1: Expected equality of these values:
1:   size
1:     Which is: 2
1:   this->eventsSize
1:     Which is: 1
1:
1:   File:     /home/fprime/Svc/BufferLogger/test/ut/Errors.cpp
1:   Line:     177
1:   Value:    Total size of all event histories
1:   Expected: 2
1:   Actual:   1
1:
1: /home/fprime/build-fprime-automatic-native-ut/F-Prime/Svc/BufferLogger/GTestBase.cpp:342: Failure
1: Expected: (this->eventHistory_BL_LogFileValidationError->size()) > (__index), actual: 0 vs 0
1:
1:   File:     /home/fprime/Svc/BufferLogger/test/ut/Errors.cpp
1:   Line:     191
1:   Value:    Index into history of event BL_LogFileValidationError
1:   Expected: Less than size of history (0)
1:   Actual:   0
1:
1: [  FAILED  ] TestErrors.LogFileValidation (10 ms)
1: [----------] 3 tests from TestErrors (29 ms total)
1:
1: [----------] 4 tests from TestLogging
1: [ RUN      ] TestLogging.BufferSendIn
1: [       OK ] TestLogging.BufferSendIn (12 ms)
1: [ RUN      ] TestLogging.CloseFile
1: [       OK ] TestLogging.CloseFile (12 ms)
1: [ RUN      ] TestLogging.ComIn
1: [       OK ] TestLogging.ComIn (11 ms)
1: [ RUN      ] TestLogging.OnOff
1: [       OK ] TestLogging.OnOff (15 ms)
1: [----------] 4 tests from TestLogging (52 ms total)
1:
1: [----------] 1 test from TestHealth
1: [ RUN      ] TestHealth.Ping
1: [       OK ] TestHealth.Ping (4 ms)
1: [----------] 1 test from TestHealth (4 ms total)
1:
1: [----------] Global test environment tear-down
1: [==========] 9 tests from 4 test suites ran. (93 ms total)
1: [  PASSED  ] 8 tests.
1: [  FAILED  ] 1 test, listed below:
1: [  FAILED  ] TestErrors.LogFileValidation
1:
1:  1 FAILED TEST
1/1 Test #1: Svc_BufferLogger_ut_exe ..........***Failed    0.12 sec
0% tests passed, 1 tests failed out of 1
Total Test time (real) =   0.13 sec
Errors while running CTest
The following tests FAILED:
          1 - Svc_BufferLogger_ut_exe (Failed)
make[3]: *** [F-Prime/Svc/BufferLogger/CMakeFiles/Svc_BufferLogger_check.dir/build.make:58: F-Prime/Svc/BufferLogger/CMakeFiles/Svc_BufferLogger_check] Error 8
make[2]: *** [CMakeFiles/Makefile2:21910: F-Prime/Svc/BufferLogger/CMakeFiles/Svc_BufferLogger_check.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:21917: F-Prime/Svc/BufferLogger/CMakeFiles/Svc_BufferLogger_check.dir/rule] Error 2
```

LeStarch, 2022-04-21:
Can you post the docker file you are using?  My suspicion is it has to do with file permissions...

Blabla51, 2022-04-22:
Here is my Dockerfile:

```
# syntax=docker/dockerfile:1
FROM ubuntu:20.04
ENV DEBIAN_FRONTEND=noninteractive
RUN apt update
RUN apt install -y make cmake gcc g++ pip git openjdk-17-jdk
RUN pip install fprime-tools==3.0.1

WORKDIR "/home"
RUN git clone https://github.com/nasa/fprime.git

WORKDIR "/home/fprime"
RUN git checkout v3.0.0
```

Here are the terminal commands I ran:
```
cd Svc/BufferLogger/
fprime-util generate --ut
fprime-util check -j 15
```

Blabla51, 2022-04-22:
I just gave it a try and run the `ci/tests/Ref.bash` script, and it did not show any error. However I'm struggling to figure out what's the difference with the commands I ran in my previous message

Hopefully it will help :)

LeStarch, 2022-05-03:
Turns out, the BufferLogger tests fail when you are root.  These tests are checking to ensure that a write permissions error will be trapped, but root always has write permissions.  The fix is to run these tests as a non-root user.

closed 2022-05-03 by LeStarch