nasa/fprime #510: CPU affinity of active components

DentOpt, 2021-05-04:
| | |
|:--- Ubuntu 20.04 :---|
|** F´ 1.5**| |
|** FW/Comp**|   |
---
## CPU affinity does not work on linux

## How to Reproduce

1. Using the Ref example 
2. Adding the Thread CPU affinity in the task start, in this example I set the cpuAffinity to 1 for all threads in `/Ref/Top/topolocy.cpp`:
```
// start(NATIVE_INT_TYPE identifier, NATIVE_INT_TYPE priority, 
//         NATIVE_INT_TYPE stackSize, NATIVE_INT_TYPE cpuAffinity)
    rateGroup1Comp.start(0, 120,10 * 1024,1); 
    rateGroup2Comp.start(0, 119,10 * 1024,1);
    rateGroup3Comp.start(0, 118,10 * 1024,1);
    // start driver
    blockDrv.start(0,140,10*1024,1);
    // start dispatcher
    cmdDisp.start(0,101,10*1024,1);
    // start sequencer
    cmdSeq.start(0,100,10*1024,1);
    // start telemetry
    eventLogger.start(0,98,10*1024,1);
    chanTlm.start(0,97,10*1024,1);
    prmDb.start(0,96,10*1024,1);

    fileDownlink.start(0, 100, 10*1024,1);
    fileUplink.start(0, 100, 10*1024,1);
    fileManager.start(0, 100, 10*1024,1);

    pingRcvr.start(0, 100, 10*1024,1);
```
3. Build the app (`fprime-util generate`, `fprime-util build`)
4. Execute the app `./build-fprime-automatic-native/bin/Linux/Ref`
5. Verify thread CPUs with either htop or 
    `ps -all | grep Ref` -> get process ID
    `taskset -cp --all-tasks -p PROCESS_ID`

## Expected Behavior
All threads are associated to CPU 1. 
However, currently threads are associated randomly to all CPUS. 

Looking at the base class implementation `FW/Comp/Task.cpp` the cpu affinity does not appear to be implement. 

I tried to add the affinity as thread attribute in line 52:
```
#ifdef TGT_OS_TYPE_LINUX
        cpu_set_t cpus; // Needed for CPU affinity
        //Setting CPU affinity
        CPU_ZERO(&cpus);
        CPU_SET(cpuAffinity, &cpus);
        if (stat != 0) {
            stat = pthread_attr_setaffinity_np(&att, sizeof(cpu_set_t), &cpus);
            return TASK_INVALID_PARAMS;
        }
#endif
```
but got the same result.
Any suggestions why the threads are not following the given CPU affinity?
Thanks!

timcanham, 2021-05-04:
@DentOpt For normal userspace Linux, we don't use the `cpuAffinity` argument. There is an alternate version in `Os/Posix/TaskRoot.cpp` that implements it. Users can link to this version if they are running as root.

We can use this issue to debate whether we should use the `cpuAffinity` in the regular userspace version.

DentOpt, 2021-05-05:
@Timcanham understood, thank you. 
From my feeling this option is not used by the majority of users. However, it would be useful to mention this option in the multicore section on the F' user guide [F´ On Baremetal and Muti-Core Systems](https://nasa.github.io/fprime/UsersGuide/dev/baremetal-multicore.html). 

Using taskroot.cpp seems to create severat issues
1. Currently it's only defined for linux (Cpu affinity can only be selected on linux systems)
2. Build errors int the logger:

I follow your recommendation to use linked the taskroot.cpp in Os/CmakeLists.txt
```
"${CMAKE_CURRENT_LIST_DIR}/Posix/Task.cpp" 
```
to 
```
"${CMAKE_CURRENT_LIST_DIR}/Posix/TaskRoot.cpp"
```
which currently leads to build errors in the logMsg , e.g. 
```
fprime/Os/Posix/TaskRoot.cpp:44:87: error: invalid conversion from ‘char*’ to ‘U64’ {aka ‘long unsigned int’} [-fpermissive]
   44 |             Fw::Logger::logMsg("pthread_attr_init: (%d)(%d): %s\n",stat,errno,strerror(stat));
```
Which you can solve by changing `strerror(errno)` to `reinterpret_cast<POINTER_CAST>(strerror(errno))`
```
Fw::Logger::logMsg("pthread_attr_init: (%d)(%d): %s\n",stat,errno,strerror(stat));
```
to 
```
Fw::Logger::logMsg("pthread_attr_init: (%d)(%d): %s\n",stat,errno, reinterpret_cast<POINTER_CAST>(strerror(stat)));
```

3. Once built and executed the Ref application finishes with 

```
fprime/Fw/Comp/ActiveComponentBase.cpp": Line: 70 2
FATAL 342 handled.
Exiting with segfault and core dump file.
Segmentation fault (core dumped)
```
Same happens using also when uncommenting all logger messages and in superuser mode `sudo`.

LeStarch, 2021-12-20:
We removed TaskRoot, and merged the Os::Task items into one implementation.  In posix, some features require permission.

closed 2021-12-20 by LeStarch