nasa/fprime #511: Add in Driver Timeout Architecture

LeStarch, 2021-05-04:
| | |
|:---|:---|
|**_F´ Version_**| devel |
|**_Affected Component_**| Drv |
---
## Feature Description

Currently there are no good ways to limit `send` and `recv` calls to a specific amount of time.  This can be done, but should be done in a generic way that works for all driver types that implement the `ByteStreamDriver` interface.

## Rationale

Calls to `send` and `recvPoll` should have a configurable timeout to prevent excessive time being spent in these calls.

LeStarch, 2021-05-04:
## How To Accomplish this:

This can be accomplished fairly accurately using the `select` interface, a non-blocking file descriptor, and a timer object. If done with the same structuring as the IP interface, we could implement a timeout driver class that oversees sends and recvs.

## Send Implementation Pseudocode (Assuming Non-Blocking Socket)

```c++
status send(int fd, char* bytes, int size, int timeout) {
    int rc = 0;
    int sent = 0;
    start = getTime(); // F´ time if possible
    while ((rc >= 0 and rc <= (total - sent))  or ((rc == -1) and errnno == EINTER || ... EAGAIN || ... EWOULDBLOCK ) {
          remaining_timeout = getTime() - start;
          rc = select(..., fd, ..., timeout );
          if (errno == TIMEOUT) {
              return TIMEOUT;  
          }
          rc = send(fd, bytes + sent, size - sent)
          if (rc > 0) {
              sent += rc
          }      
    }
}
```

matt392code, 2024-12-28:
Suggested Enhancements:
Driver Architecture:
```
class TimeoutByteStreamDriver : public ByteStreamDriver {
    private:
        ByteStreamDriver& wrapped_driver;
        const U32 send_timeout_ms;
        const U32 recv_timeout_ms;
        
    public:
        TimeoutByteStreamDriver(
            ByteStreamDriver& driver,
            U32 send_timeout,
            U32 recv_timeout
        );
        
        status_t send(const U8* const data, const U32 size) override;
        status_t recv(U8* const data, const U32 size) override;
};
```
Additional Features to Consider:
>Configurable timeouts per operation
>Timeout statistics tracking
Integration with F Prime's event system for timeout notifications
Mechanism to cancel pending operations

Testing Considerations:
>Unit tests for various timeout scenarios
Tests for partial sends/receives
Edge case testing (0 timeout, very large timeout)
System tests for real-world timing behavior

Configuration:
```
struct TimeoutConfig {
    U32 send_timeout_ms;  // Send operation timeout
    U32 recv_timeout_ms;  // Receive operation timeout
    bool enable_stats;    // Enable timeout statistics
    U32 retry_count;      // Number of retries on timeout
};
```

LeStarch, 2025-12-19:
Timeouts are supplied at configuration time and at the discretion of Driver implementor.