nasa/fprime #516: File Downlink File Names

LeStarch, 2021-05-05:
| | |
|:---|:---|
|**_F´ Version_**| 1.5.3 |
|**_Affected Component_**| FileDownlink  |
---
## Problem Description

Some file names reportedly break FileDownlink. 

## How to Reproduce

"e.g. a source file name of "/tmp/Screen_Shot_2020-03-26_at_9.19.28_PM.png" breaks it with format error."

Joshua-Anderson, 2021-05-06:
The default F' FW_CMD_STRING_MAX_SIZE is 40 characters. `/tmp/Screen_Shot_2020-03-26_at_9.19.28_PM.png` is 45 characters, longer the max supported string length.

I'm tempted to close this as expected behavior, but we should start a conversation about how to handle strings in F' better. For example, the command definition for FileDownlink.SendFile defines source file as a string of length 100. Should requesting a command string size longer than FW_CMD_STRING_MAX_SIZE somehow result in a warning?

timcanham, 2021-05-06:
We should make this part of a larger effort to define constants in a way that the autocoder, ground system, and software can all use them and that way we can add autocoder rules to detect issues like this.

LeStarch, 2021-05-06:
@timcanham and @Joshua-Anderson, I agree with both here.  Closing in favor of: https://github.com/nasa/fprime/issues/530

closed 2021-05-06 by LeStarch