nasa/fprime #519: GDS: Double Click Repopulation Not Working

LeStarch, 2021-05-05:
| | |
|:---|:---|
|**_F´ Version_**| 1.5.3/devel |
|**_Affected Component_**|  GDS GUI  |
---
## Problem Description

Double clicking on commands in the command history should repopulate the command-send box.

## How to Reproduce

1.  Run GDS/GUI with Ref
2. Send command NO-OP STRING with string arg
3. Send command NO-OP without string
4. Double click on the command history entry for NO-OP-STRING.
5. It **should** repopulate the send box.

## Expected Behavior

Double clicking on a command hisotry item should repopulate the command send box with that command.

LeStarch, 2021-05-05:
@saba-ja if you have time could you look into this?

saba-ja, 2021-05-06:
To clarify, the expectation is to click on a row in CMD hist table and make it to be selected in the Mnemonic dropdown. Also if it has args populate them in the arg input boxes? 

[image]

Is this an expected behavior or a feature request?
I do not see any implementation of `clickAction` shown below:
https://github.com/nasa/fprime/blob/97883e50982a1275383a721f6c8a1c4c2aa1433e/Gds/src/fprime_gds/flask/static/index.html#L344-L349

in command.js
https://github.com/nasa/fprime/blob/update/ground-prototype/Gds/src/fprime_gds/flask/static/js/vue-support/command.js

@LeStarch, Please advice if this functionality should be implemented.

Joshua-Anderson, 2021-05-06:
@saba-ja Exactly! You should be able to click on a row in the command history and re-populate it back up at the top. Makes it easy to re-send previously sent commands.

LeStarch, 2021-05-06:
@saba-ja the implementation for `clickAction` probably got moved or broke at some point.  This is likely why it stopped working.

Joshua-Anderson, 2021-05-25:
Fixed in #604!

closed 2021-05-25 by Joshua-Anderson