nasa/fprime #520: U8* Items Aren't Handled Well

LeStarch, 2021-05-05:
| | |
|:---|:---|
|**_F´ Version_**| devel |
|**_Affected Component_**|  U8* autocoded output  |
---
## Problem Description

Using a U8* as an argument to a port causes bad autocoding in UTs.  Using a U8* in a complex type breaks.  We should have a consistent approach to U8*.

bocchino, 2021-05-14:
Note that FPP models currently have no way to represent pointer types. If you want to pass a pointer across a port or store one in a struct or array, you wrap it in a smart pointer or use POINTER_CAST. So adding this kind of support for raw pointers would be an enhancement to FPP as well.

bocchino, 2021-05-14:
I might say this is not a bug, but a feature request/enhancement.

bocchino, 2021-05-14:
...since we do have a way to represent pointers (POINTER_CAST or wrapping) just not as a raw U8*.

LeStarch, 2021-05-14:
I am going to close this issue in this case.  We can move forward with the above suggested approaches:

1. Wrap U8* in a complex type (Serializable)
2. Cast as POINTER_CAST
3. Use an array type.

LeStarch, 2021-05-14:
If there are more concerns please reopen the issue.

closed 2021-05-14 by LeStarch