nasa/fprime #521: Clarify fprime-gds requirements

BastianZim, 2021-05-05:
Hi,

currently, fprime-gds has multiple sources of mandatory and optional requirements.

It would be good to know which of those are the authoritative ones.

Examples:

- https://github.com/nasa/fprime/tree/devel/Gds#setup
- https://nasa.github.io/fprime/INSTALL.html
- https://github.com/nasa/fprime/blob/7219f13ec9fdf2172eb349582bd0601c35a3ffc2/Gds/setup.py#L110-L123
- https://github.com/nasa/fprime/blob/devel/Gds/requirements.txt

BastianZim, 2021-05-05:
Happy to create a PR but just wanted to make sure what to in- and exclude.

BastianZim, 2021-05-05:
Additionally, _flask_uploads_ is fixed to a commit. Is there a specific reason why it's not a version?

Joshua-Anderson, 2021-05-05:
It looks like flask_uploads hasn't been published to pypi in some time, so that's probably why it's pinned to a specific commit. 

The GDS readme is outdated, wxgui GDS support has been removed in favor of the http GDS.
The INSTALL instructions are our source of truth, how do they disagree with the GDS pip requirements? We have a step to update pip and setuptools before installing python dependencies because a lot of python3.6 distributions ship a version of pip and setuptools is so outdated they cannot be used to directly install the python dependencies.

BastianZim, 2021-05-05:
> It looks like flask_uploads hasn't been published to pypi in some time, so that's probably why it's pinned to a specific commit.

Oh yes you're right: https://github.com/maxcountryman/flask-uploads/issues/41 Ok makes sense.

> The GDS readme is outdated, wxgui GDS support has been removed in favor of the http GDS.

Should that be removed, or do you want to keep that just in case?

> The INSTALL instructions are our source of truth, how do they disagree with the GDS pip requirements?

They agree, just wanted to include it for completeness. :)

Joshua-Anderson, 2021-05-05:
Yeah, we should remove references to the old WX gui to avoid confusion like you experienced!

BastianZim, 2021-05-05:
Awesome, will try to create a PR.

LeStarch, 2021-05-05:
@BastianZim awesome, thanks!

PR #524 (2021-05-06): Remove unused dependencies from GDS
| | |
|:---|:---|
|**_Originating Project/Creator_**| |
|**_Affected Component_**| GDS |
|**_Affected Architectures(s)_**|  |
|**_Related Issue(s)_**| Closes #521 |
|**_Has Unit Tests (y/n)_**| n |
|**_Builds Without Errors (y/n)_**|  |
|**_Unit Tests Pass (y/n)_**|  |
|**_Documentation Included (y/n)_**| n |

---
## Change Description

A description of the changes contained in the PR.

Removes unused dependencies 

## Rationale

A rationale for this change. e.g. fixes bug, or most projects need XYZ feature.

Decluttering

## Testing/Review Recommendations

 [... cut]
closed 2021-05-06 by LeStarch