nasa/fprime #534: installation guide not working

Joel-Nickson, 2021-05-08:
| | |
|:---|:---|
|**_F´ Version_**| @latest |
|**_Affected Component_**|  install.html |
---
## installation guide not entirely working on windows wsl

Following https://nasa.github.io/fprime/INSTALL.html fprime-util install doesn't seem to work right.

This is the error msg received on `frime-util generate`:

[WARNING] Failed to find settings file: /mnt/d/Projj/nasa/helicopter2021/Fprime/fprime/Ref/settings.ini
[ERROR] /mnt/d/Projj/nasa/helicopter2021/Fprime/fprime/Ref/build-fprime-automatic-native already exists.

This is the error msg received on `fprime-util install --jobs 32`:

usage: fprime-util [-h] {generate,purge,hash-to-file,info,build,impl,check} ...
fprime-util: error: argument command: invalid choice: 'install' (choose from 'generate', 'purge', 'hash-to-file', 'info', 'build', 'impl', 'check')

## How to Reproduce

1. run wsl on windows
2. follow the installation guide.

## Expected Behavior

Not working

LeStarch, 2021-05-09:
This is a divergence between the publish guide matching `master`/version `1.5.3` and the `devel` branch.  We removed `install` from the devel branch as  the "build" step now does the install.

The latest INSTALL.md is found here: https://github.com/nasa/fprime/blob/devel/docs/INSTALL.md

closed 2021-05-09 by LeStarch

Joel-Nickson, 2021-05-09:
thanks 

but I  do believe that more people who want to check F' would have the same problem as mine , so maybe providing the above link over there would be a lot helpful