nasa/fprime #549: Improve type safety of Fw::Logger

bocchino, 2021-05-13:
| | |
|:---|:---|
|**_F´ Version_**| v1.10 |
|**_Affected Component_**| All |
---
## Feature Description

The Fw::Logger interface requires that arguments of type POINTER_CAST be bound to format specifiers. 

https://github.com/nasa/fprime/blob/67b7a7d7177af832c02893bac5db2a22a4b8e7fe/Fw/Logger/Logger.hpp#L54-L66

Therefore using the logger like this
```
Fw::Logger::logMsg("%u %u", 0, 1);
```
assumes that code like this works properly:
```
#include <stdint.h>
#include <stdio.h>
 
int main(int argc, char **argv) {
  const uint64_t a = 0;
  const uint64_t b = 1;
  printf("%u %u\n", a, b);
  return 0;
}
```

This code does work on my machine (Mac OS 10.15, clang), but (a) it produces warnings about invalid format specifiers and (b) according to the C library standard it is not guaranteed to work. See https://en.cppreference.com/w/c/io/fprintf. 

The only absolute guarantee is that this code works:
```
#include <stdio.h>
 
int main(int argc, char **argv) {
  const unsigned a = 0;
  const unsigned b = 1;
  printf("%u %u\n", a, b);
  return 0;
}
```

If you know that uint32_t is an alias for unsigned, then this also works:
```
#include <stdint.h>
#include <stdio.h>
 
int main(int argc, char **argv) {
  const uint32_t a = 0;
  const uint32_t b = 1;
  printf("%u %u\n", a, b);
  return 0;
}
```
However, this version is incorrect on platforms where unsigned is an alias for uint16_t.

A safer approach would be to use `Fw::FixedLengthString::format` to format the string before passing it to the logger:
```
Fw::FixedLengthString s;
s.format("%u %u\n", 0, 1);
Logger::logMsg(s.toChar());
```
This way the correct argument types are bound to the correct format specifiers, without passing through POINTER_CAST.

## Rationale

In general we should avoid code that causes compiler warnings and/or relies on non-standard behavior. In this case, the compiler and library writers have clearly done something nifty to make printf "just work" despite the mismatched argument types, but we may not want to rely on this nifty behavior in the core framework.

zachar1a, 2021-08-07:
### using macOS BigSur 11.5
I've tried to work on this issue but I wanted to see how `Fw::FixedLengthString` works so I created a test and defined it as:
[image]

And have been receiving this error when trying to run the tests using `fprime-util check --all`:
[image]

I just want to make sure that it's me messing something up in the process before anything is changed because im not sure this would be a priority. 

Also I noticed in the Fw/Types/test/ut/TypesTest.cpp file there are 3 `#include <iostream>`

bocchino, 2021-08-09:
`Fw::FixedLengthString` is not enabled on the `devel` branch. To use it, you will need to add it to `Fw/Types/CMakeLists.txt`.

There is an open PR against release/v3.0.0 to rename `Fw::FixedLengthString` to `Fw::String` and replace all uses of `Fw::EightyCharString` with `Fw::String`. See https://github.com/nasa/fprime/pull/893.

zachar1a, 2021-08-09:
Ah ok thank you.

LeStarch, 2021-12-20:
Closing this in favor of #1055.

closed 2021-12-20 by LeStarch