nasa/fprime #550: Disable LENGTH

LeStarch, 2021-05-14:
| | |
|:---|:---|
|**_F´ Version_**| devel |
|**_Affected Component_**| XML Enums |
---
## Discussion Question

We added _LENGTH as a constant to help with autocoded enumerations.  The rationale was that it is nice to be able to bounds-check and iterate over enumeration values.

However, it is left to the user to understand if the enumerations are sequential and zero-based when using this field.  The discussion question at hand is this:

In cases where the values of the enumeration are non-sequential and not-zero based, should this field be disabled or left up to the user?

Joshua-Anderson, 2021-05-14:
Since it's now called `NUM_CONSTANTS` the name is now based on the number of elements in the enumeration and doesn't imply iterablity, so I personally don't think we need to disable it if the elements aren't sequential.

LeStarch, 2021-05-14:
Good call!  I'll leave this open for others to comment, but I think that is the answer.

bocchino, 2021-05-14:
I agree. I can't think of any valid use for `NUM_CONSTANTS` if the constants are not dense array indices, but I guess it doesn't hurt to leave it there in case someone wants to use it.

bocchino, 2021-05-14:
OK, here's a possible valid use. Before a switch statement that checks all cases, you can assert that the number of constants is equal to the number of branches in the switch. That would provide assurance that you've covered all the cases. If not, the assertion would fail whenever the switch runs. The current practice is to add a default with `FW_ASSERT(0)`. That fails only if the missing branch is actually taken.

LeStarch, 2021-05-27:
Ok, I am closing this issue!

closed 2021-05-27 by LeStarch