nasa/fprime #568: Add Ability to Specify Telemetry Conversions in Telemetry Dictionary

arizvi786, 2021-05-18:
Currently telemetry conversions for raw values output in telemetry can't be specified in the telemetry dictionary. Having the ability to specify telemetry conversions in the dictionary would be good feature to add. Also, having this feature be compatible with other commonly used GDS systems would be more beneficial.

Here's an example telemetry conversion specification: -

<telemetry abbreviation=”A-9999” name=”Polynomial_EU_Example” type=”float” byte_length=”4”>
      <raw_to_eng>
            <polynomial_expansion>
                  <factor index=”0” coeff=”1.2” />
                  <factor index=”1” coeff=”0.75” />
                  <factor index=”2” coeff=”2.2” />
            </polynomial_expansion>
      </raw_to_eng>
</telemetry>

timcanham, 2021-05-18:
How about a generic converter plug-in concept? Then the user could write their own. The polynomial converter could be one of the standard ones. Maybe:

```xml
<telemetry abbreviation=”A-9999” name=”Polynomial_EU_Example” type=”float” byte_length=”4”>
      <converter name="poly">
            <args>
                  <factor index=”0” coeff=”1.2” />
                  <factor index=”1” coeff=”0.75” />
                  <factor index=”2” coeff=”2.2” />
            </args>
      </converter>
</telemetry>

The `args` could be passed as named arguments to a `convert` proc in the converter class.

arizvi786, 2021-05-18:
That's a good idea. The only thing there is we'd have to translate this format to another that's compatible with another GDS if we're using F-Prime with that other GDS.

matt392code, 2024-12-28:
Proposed solution that builds on both suggestions while addressing the compatibility concern:
[telemetry-converter.py.txt](https://github.com/user-attachments/files/18264850/telemetry-converter.py.txt)

This implementation provides several key features:
1. A plugin-based converter system as suggested by timcanham
2. Support for converting configurations to other GDS formats (like ITOS and YAMCS) addressing arizvi786's compatibility concern
3. A flexible architecture that allows for:
   - Easy addition of new converter types
   - Support for additional GDS format translations
   - Clear separation of concerns between conversion logic and format handling

The solution uses:
1. Abstract base classes to define the converter interface
2. A factory pattern for creating converters
3. XML parsing for configuration handling
4. NumPy for efficient polynomial calculations

To add support for a new converter type, you would:
1. Create a new class inheriting from TelemetryConverter
2. Implement the convert() and to_gds_format() methods
3. Register it with the ConverterFactory

To add support for a new GDS system:
1. Add a new format option in the to_gds_format() method
2. Implement the corresponding format conversion method

thomas-bc, 2025-06-11:
This could be (partly?) addressed by https://github.com/nasa/fprime/issues/3509 ?

closed 2025-07-29 by thomas-bc