nasa/fprime #602: GDS does not support display of complex type as argument

ugo94490, 2021-05-21:
| | |
|:---|:---|
|**_F´ Version_**|**devel**|
|**_Affected Component_**|**GDS**|
---
## Problem Description

According to @lestarch [[1]](https://groups.google.com/g/fprime-community/c/yf6NMXk0NKM) actually the GDS does not support complex types such as arrays or serializables, however no error is detected and so the GDS is launched which causes a white screen.

The GDS shall be able to take into account these different types.

## How to Reproduce

1. Define a complex type

```xml
<array name="Quaternion" namespace="Ref">
    <type>F64</type>
    <size>4</size>
    <format>%f</format>
    <default>
        <value>0</value>
        <value>0</value>
        <value>0</value>
        <value>0</value>
    </default>
</array>
```

2. Define a component command using this complex type in argument

```xml
<import_array_type>Ref/TCModule/QuaternionArrayAi.xml</import_array_type>
<command kind="async" opcode="1" mnemonic="TC_Tracking">
    <comment>
        Command to switch in Tracking Mode
    </comment>
    <args>
        <arg name="arg1" type="Ref::Quaternion"/>
    </args>
</command>
```

3. Launch fprime build, everything nominal

4. Launch GDS

GDS does not display anymore: white screen and we this error in the console:

``TypeError: command.args is undefined``

## Expected Behavior

This feature shall be implemented as it could facilitate the sending of vector or linked data set.

While waiting for this bug to be solved from our point of view there are three temporary solutions to notify the developer of this bug:

1. Notify the error in the fprime-util generate command
2. Notify the error when launching fprime-gds
3. Notify the error in the web browser via a pop-up

ThibFrgsGmz, 2021-05-24:
This bug was previously reported in issue #322 and was closed by @LeStarch with a suggestion to open an enhancement issue.

Can you take the action to open an enhancement issue or modify the current issue?

ugo94490, 2021-05-25:
This is more of an enhancement than a bug so i reopen an issue : [#607](https://github.com/nasa/fprime/issues/607)

closed 2021-05-25 by ugo94490