nasa/fprime #678: Resolve Static Analysis Warnings

saba-ja, 2021-06-07:
| | |
|:---|:---|
|**_F´ Version_**| 1.5 |
|**_Affected Component_**| All  |
---
## Problem Description
There are 444 warnings after running CodeSonar on Ref deployment.
This ticket will be used to track series of PRs resolving these warnings.

Category and significance  of the warnings is listed below:

| significance | count |
| -- | -- |
|Security | 399 |
|Reliability | 10 |
|Redundancy | 35 |

Class | Count
| -- | -- |
Uninitialized Variable | 255
Cast Alters Value | 120
Unreachable Data Flow | 15
Redundant Condition | 11
No Space For Null Terminator | 9
File System Race Condition | 7
Useless Assignment | 7
Coercion Alters Value | 6
Ignored Return Value | 6
Null Pointer Dereference | 3
Integer Overflow of Allocation Size | 1
Dangerous Function Cast | 1
Tainted Buffer Access | 1
Unreachable Computation | 1
Unreachable Conditional | 1

## How to Reproduce

1. Run CodeSonar on Ref deployment

## Expected Behavior

Warnings should be either resolved or descoped with a rational.

LeStarch, 2022-09-30:
As part of this we need to run static analysis in CI and continue to fix.

matt392code, 2024-12-28:
Create a C++ static analysis helper class that can be used alongside the existing F´ framework.
[static-analysis-helper.txt](https://github.com/user-attachments/files/18265520/static-analysis-helper.txt)
Example usage code to demonstrate how to use these utilities:
[static-analysis-example.txt](https://github.com/user-attachments/files/18265521/static-analysis-example.txt)
These utilities provide several safety features:
1. BLOCKER (Security/Null Pointers):
- Pointer validation before use
- Assert-based pointer checking
- Templated pointer safety checks

2. CRITICAL (Buffer/Initialization):
- Buffer size validation
- String safety checks with null termination verification
- Initialization tracking for critical variables
- Buffer overflow prevention

3. MAJOR (Cast/Coercion):
- Type-safe casting with range checking
- Signed/unsigned conversion safety
- Floating-point to integer safety checks
- Compile-time type validation

Key Features:
1. Template-based to work with any data type
2. Compile-time checks where possible
3. Runtime assertions for dynamic checks
4. Easy to integrate with existing F´ components

matt392code, 2024-12-28:
MINOR: Redundancy warnings
Code to help detect and prevent redundancy issues commonly found in static analysis.
[redundancy-checker.txt](https://github.com/user-attachments/files/18265526/redundancy-checker.txt)
An example showing how to use these redundancy checkers:
[redundancy-example.txt](https://github.com/user-attachments/files/18265529/redundancy-example.txt)
This code helps detect and prevent several types of redundancy warnings:
1. Redundant Conditions:
   - Tracks and detects repeated condition checks
   - Helps eliminate duplicate if statements
   - Reduces cyclomatic complexity
2. Dead Code Detection:
   - Tracks variable access and modification
   - Identifies unused modified values
   - Helps find unreachable code paths
3. Unused Parameters:
   - Tracks parameter usage in functions
   - Identifies unused function parameters
   - Helps maintain clean interfaces
4. Redundant Operations:
   - Detects redundant bit operations
   - Identifies unnecessary memory operations
   - Checks for redundant string operations
5. Memory Operation Optimization:
   - Prevents unnecessary memory copies
   - Detects redundant memory initialization
   - Optimizes string operations