nasa/fprime #680: Typo in GPS Tutorial.

rawinza555, 2021-06-07:
| | |
|:---|:---|
|**_F´ Version_**| |devel
|**_Affected Component_**| GpsComponentImpl.cpp |
---
## Problem Description

The function used in GpsComponentImpl.cpp in GPS Tutorial is not correct. It should be setData, setSize and getData based on what is in FW/Buffer/Buffer.cpp but you have it in all lowered case in the example code in the GPS tutorial. Not sure if you guys modify buffer.cpp and forget to change the tutorial.

## How to Reproduce

Follow the gps tutorial until you build the gps component after you modify GpsComponentImpl.cpp and GpsComponentImpl.hpp

## Expected Behavior

It should show an error that setdata, setsize and getdata are not in a class buffer.

LeStarch, 2021-06-07:
This is absolutely the case, we have updated Fw buffer to follow naming conventions without changing the GPS tutorial.  If you'd like to file a PR go ahead and do so, otherwise this change will be reflected soon.

LeStarch, 2021-12-20:
I think this was done, although that tutorial should be updated again.  That ticket can be posted on that repo.

closed 2021-12-20 by LeStarch