nasa/fprime #684: No error message for conflicting event IDs

aidan-wagner, 2021-06-07:
| | |
|:---|:---|
|**_F´ Version_**| Devel |
|**_Affected Component_**| fprime-util |
---
## Feature Description

Currently, if two events are created with the same ID, there is no description of this error provided to the developer. It would be helpful to create an error message to guide the developer in the direction of this error when attempting to run fprime-util build.

## Rationale

This bug can be difficult to locate right now because the message in the terminal provides no description as to where the error is or what is causing the error.

Joshua-Anderson, 2021-06-07:
@aidan-wagner If you still have it, can you print the error message output you received?

LeStarch, 2021-12-20:
Should be caught by FPP.

closed 2021-12-20 by LeStarch