nasa/fprime #709: docs: INSTALL.md: Unable to test F` GDS installation via running HTML GUI

FedorChervyakov, 2021-06-14:
| | |
|:---|:---|
|**_F´ Version_**| 2.0.0|
|**_Affected Component_**|   docs, GDS|
---
## Problem Description

I am new to the F\`. I am trying to follow the installation guide [INSTALL.md](https://github.com/nasa/fprime/blob/devel/docs/INSTALL.md). Everything works fine until the **Testing F´ GDS Installation Via Running HTML GUI** step. When running the `fprime-gds -g html -r ./build-artifacts` from the Ref directory I experience the following error:
```
[ERROR] Dictionary version 2.0.0 is not in supported range: 0.0.0-1.5.4. Please upgrade fprime-gds.
[ERROR] Failed to run application: ****/fprime-venv/bin/python3 -u -m flask run --host 127.0.0.1 --port 5000. Error: ****/fprime-venv/bin/python3 stopped with code 1 sooner than 2 seconds.
```
<details>
<summary>Expand to view full log</summary>

```
» fprime-gds -g html -r ./build-artifacts
[INFO] Ensuring TCP Server is stable for at least 5 seconds
[INFO] Running Application: ****/fprime-venv/bin/python3
[INFO] Log File: ****/fprime/Ref/logs/2021_06_14-04_30_40/ThreadedTCP.log
[INFO] Ensuring comm[ip] Application is stable for at least 1 seconds
[INFO] Running Application: /home/fedor/packages/fprime-venv/bin/python3
[INFO] Ensuring Ref Application is stable for at least 1 seconds
[INFO] Running Application: ****/fprime/Ref/build-artifacts/Linux/bin/Ref
[INFO] Log File: ****/fprime/Ref/logs/2021_06_14-04_30_40/Ref.log
[INFO] Ensuring HTML GUI is stable for at least 2 seconds
[INFO] Running Application: ****/fprime-venv/bin/python3
 * Serving Flask app "fprime_gds.flask.app"
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
[ERROR] Dictionary version 2.0.0 is not in supported range: 0.0.0-1.5.4. Please upgrade fprime-gds.
[ERROR] Failed to run application: ****/fprime-venv/bin/python3 -u -m flask run --host 127.0.0.1 --port 5000. Error: ****/fprime-venv/bin/python3 stopped with code 1 sooner than 2 seconds.
[INFO] Shutting down F prime due to error. Failed to run HTML GUI
```
</details>

## How to Reproduce
I am following the [INSTALL.md](https://github.com/nasa/fprime/blob/devel/docs/INSTALL.md) verbatim:
```
python3 -m venv ./fprime-venv
./fprime-venv/bin/activate
git clone https://github.com/nasa/fprime
cd fprime
cd Ref
pip install --upgrade fprime-tools fprime-gds
fprime-util generate\nfprime-util build --jobs "$(nproc || printf '%s\n' 1)"
fprime-gds -g html -r ./build-artifacts
```

## Expected Behavior

Joshua-Anderson, 2021-06-14:
@LeStarch Did you publish the v2.0.0 versions of the tools to pypi?

LeStarch, 2021-06-14:
Not yet.  This will be done shortly and will resolve this issue.

LeStarch, 2021-06-14:
This is fixed with the latest fprime-gds release.

```
pip install -U fprime-gds
```

We have also taken release process improvements to try and mitigate this effect in the future.

LeStarch, 2021-06-14:
Note:  you'll need fprime-gds version 2.0

LeStarch, 2021-06-14:
I am going to close this.  Feel free to reopen if there is still issue here.

closed 2021-06-14 by LeStarch

FedorChervyakov, 2021-06-15:
Everything works fine after upgrade. Thank you.