nasa/fprime #722: Support passing by const reference

Joshua-Anderson, 2021-06-16:
| | |
|:---|:---|
|**_F´ Version_**| latest |
|**_Affected Component_**|  autocoder |
---
## Feature Description

Right now, there is no option in the autocoder to pass by const reference. This makes it impossible to use const references as arguments to port calls. This can be worked around by using const casts to remove the constness of the reference, but this defeats the point of const correctness throughout F'.

Joshua-Anderson, 2021-06-16:
@bocchino Since you're working on F'', I was wondering if you have already been thinking about this. Do you have any plans for supporting const references within F''?

bocchino, 2021-06-16:
We could add support for this in the FPP model if necessary. However, my sense is that 'const reference' is a lower-level concept than we want to support in the FPP model.

In the FPP model, the main distinction is between "pass by mutable reference" (i.e., allowing the receiver to update a value held by the sender) and "pass by value" (i.e., not allowing that). In the context of a synchronous port invocation, "pass by const reference" is an optimization that provides pass-by-value semantics and avoids extra copies. I feel that this kind of copy-avoidance can be done in the code generator, by inserting const references automatically where appropriate.

bocchino, 2021-06-16:
A related point is that const references don't play well with F Prime serializable values, as they currently work. In F Prime, typically what we pass over a port is a small scalar value or a serializable class instance. A small scalar value is OK to pass by value or by mutable reference; const references are not needed. A serializable class instance can't be passed by const reference, because it's a container containing data and metadata, and to do anything with the data (serialize or deserialize it) the receiver needs to update the metadata.

I think if we wanted to pass serializable values by const reference, we would need to extend the semantics of serializable values at the framework level. We would need something like a "constant deserialization handle" that provides a serializable view over here into data stored over there, and supports only deserialization.

bocchino, 2021-06-17:
> I think if we wanted to pass serializable values by const reference, we would need to extend the semantics of serializable values at the framework level. We would need something like a "constant [...]

This comment applies to serialized buffers (e.g., Fw::Com), but not to serialized class instances with members (e.g., Fw::Time). You can use a const reference to read the members.

bocchino, 2021-07-09:
So it turns out that const references can be used for serial buffers, if you want to copy the whole buffer. Where const references don't work is if you want to pull individual items out of the buffer, because that requires updating the metadata.

@rohandkn Is working on the optimization to implement "pass by value" using const references where appropriate.

matt392code, 2024-12-31:
[const-ref-support.txt](https://github.com/user-attachments/files/18279868/const-ref-support.txt)
This solution provides:
1.  Smart Type Selection
>Automatically determines when const ref is beneficial
>Avoids overhead for small types
>Handles fundamental types appropriately
2.  Specialized Serialization
3.  Flexible Port Arguments
>Wraps arguments appropriately
>Maintains type safety
>Supports both const ref and value semantics
4.  Key Benefits:
>No unnecessary copying
>Maintains const correctness
>Handles both buffer and non-buffer types
>Compatible with F Prime's serialization system

closed 2025-01-07 by bocchino