nasa/fprime #764: Update Schematron check for active components

timcanham, 2021-06-24:
| | |
|:---|:---|
|**_F´ Version_**| v2.0.0 |
|**_Affected Component_**|   |
---
## Problem Description

The schematron check fails for the case where there is an active component with only a sync port but with async commands.

## How to Reproduce

1. Create an active component with 1 async command and one sync port
2. Compile XML

## Expected Behavior

This is a valid combination, so the schematron check should pass.

timcanham, 2021-06-24:
Diff for tested modification:

```diff
diff --git a/Autocoders/Python/schema/default/active_comp_schematron.rng b/Autocoders/Python/schema/default/active_comp_schematron.rng
index 38df16230..8eaeb5bb7 100644
--- a/Autocoders/Python/schema/default/active_comp_schematron.rng
+++ b/Autocoders/Python/schema/default/active_comp_schematron.rng
@@ -3,7 +3,7 @@
         xmlns="http://purl.oclc.org/dsdl/schematron">
     <pattern id="ASYNC_PORT">
         <rule context="component">
-            <assert test="(//component/@kind = 'active' and (count(//component/ports/port/@kind[. = 'async_input']) + count(//component/internal_interfaces/internal_interface) ) >= 1) or not (//component/@kind = 'active')">
+            <assert test="(//component/@kind = 'active' and (count(//component/ports/port/@kind[. = 'async_input']) + count(//component/commands/command/@kind[. = 'async']) + count(//component/internal_interfaces/internal_interface) ) >= 1) or not (//component/@kind = 'active')">
                 Active components should have at least 1 port of kind async_input or internal_interface.
             </assert>
         </rule>
```

timcanham, 2021-06-24:
Note that this case should check for an `internal_interface` as well, since that would also be valid.

LeStarch, 2021-12-20:
Done already.

closed 2021-12-20 by LeStarch