nasa/fprime #767: fprime-util purge function for build file that is not from a default.

rawinza555, 2021-06-24:
| | |
|:---|:---|
|**_F´ Version_**| 1.5|
|**_Affected Component_**| fprime-util purge |
---
## Feature Description

When `fprime-util purge` is executed, it should also ask us if we want to delete the build file that is not from a default setting. Let's say if I have `build-fprime-automatic-native` and `build-fprime-automatic-Raspberrypi`, by running `fprime-util purge`, it should also try to delete `build-fprime-automatic-Raspberrypi` as well. The current version would just delete the native one.

## Rationale

It would make deleting and building new version of the app/project easier.

Joshua-Anderson, 2021-06-24:
Makes sense to me. There's not a lot of times I'd want to purge for one platform and not the rest. For those exceptions where you do only want to purge one platform you can always answer no on the prompt for the other platforms. What are everyone else's thoughts?

LeStarch, 2021-12-20:
This works if you supply the toolchain.  Since the pattern to work with non-default toolchains is to supply them my thoughts are that purge should also take this input.  Otherwise `purge` falls out of pattern with the others and is the odd-one-out.

Closing this issue, further discussion is welcomed here: https://github.com/fprime-community/fprime-tools

closed 2021-12-20 by LeStarch