nasa/fprime #773: Autocoder does not export full path to topology.xml

DentOpt, 2021-06-25:
| | |
|:---|:---|
|**_F´ Version 1.5**| |
|**_MagicDraw Autocoder_**|   |
---
## Problem Description

Autocoder does not export full path to topology.xml

## Expected Behavior
While e.g. the path of component `ManagerMission` should be `ManagerMission/ManagerMissionComponentAi.xml`, 
it always exports the path as `/ManagerMissionComponentAi.xml` in the topolog.xml. 
Is there anyway to set this export path in Magicdraw?

## How to Reproduce

The repo configuration:

I have the following repository setup, with the MagicDraw project file at the highest level. 
```
├── Project.mdxml  (THIS IS THE MAGICDRAW PROJECT FOR THE OVERALL SYSTEM)
├── CMakeLists.txt
├── settings.ini   (project_root: .  framework_path: ./Fprime)
├── Fprime
│   ├── Autocoders
│   │   ├── CMakeLists.txt
│   │   ├── MagicDrawCompPlugin
│   
│     ...
│
├── ManagerMission
│   ├── CMakeLists.txt
│   ├── ManagerMissionComponentAi.xml
│   └── ManagerMission.mdxml (THIS IS THE MAGICDRAW PROJECT FOR THE COMPONENT)
│   
└── Topology
    ├── CMakeLists.txt
    ├── Main.cpp
    ├── README.md
    ├── RoboticsProcessingUnitTopologyAppAi.xml (THIS IS WHAT WE WANT TO GENERATE)
    └── Topology.cpp
```    
The exported topology looks like this:
```    
<assembly name = "RoboticsProcessingUnit">
	<import_component_type>Svc/TlmChan/TlmChanComponentAi.xml</import_component_type>
	<import_component_type>/ManagerMissionComponentAi.xml</import_component_type>  
```    
The MagicDraw project setup:
[image]

## What I tried

I try to understand how the autocoder is exporting the component path as apparently for the Ref Application this is just working fine. I currently assume that the export is based on the "preferred path value ". However I cannot modify this value directly. I tried: 
- Changing the preferred path value directly in the Project.mdxml
- Using different sharing folder structures in ManagerMission.mdxml: 
```
Model
└── ManagerMission (Shared Package)
          └── ManagerMission (Queued passive component)
Model
└── ManagerMission (Shared Package)
         └── ManagerMission (Package) 
                         └── ManagerMission (Queued passive component)
Model
└── ManagerMission
         └── ManagerMission  (Shared Package)
                         └── ManagerMission (Queued passive component)
- Adding "ManagerMission"  as namespace in queued passive component
- Modifying preferred path in package option dialog (not possible)
- Modifying Mounted on in package option dialog (does lead to different project structure and not impact exported path
````

[image]

Joshua-Anderson, 2021-06-25:
Hmmm, we've slowly been deprecating magicdraw support, but this doesn't seem like something that should have been broken. Maybe somebody with magicdraw experience can chime in with some hints.

LeStarch, 2021-07-07:
Are you setting the namespace setting in the magic draw component?

LeStarch, 2021-12-20:
Magic draw support is deprecated, and this issue has not been followed up.  Feel free to ask a question using discussions if more is needed.

closed 2021-12-20 by LeStarch