nasa/fprime #805: Buffer.cpp assert when constructor is called with NULL pointer

timcanham, 2021-06-30:
| | |
|:---|:---|
|**_F´ Version_**|2.0.0 |
|**_Affected Component_**| various  |
---
## Problem Description

If a user creates an `Fw::Buffer` instance this way:

```
Fw::Buffer fwBuffer(NULL, 0, 0);
```

The code will assert. 

## How to Reproduce

## Expected Behavior

Debate: Should NULL be a valid value? If so, this should be fixed.

Joshua-Anderson, 2021-06-30:
I'd argue that even if 0 is a valid memory address on your system, you really shouldn't use it because you're just asking for trouble. As I've learned the hard way when building my own toy OS for a class, trying to use address 0 quickly becomes a nightmare because you lose the ability to determine if a function returned an error or a pointer to the address 0.

I think the buffer asserting when given a null pointer is a good safety feature because it catch cases when memory allocators like malloc and new return an error (NULL), even if a user explicitly forgot to check the return value.

LeStarch, 2021-07-07:
@timcanham is your desire to treat "0" as a valid address like @Joshua-Anderson is suggesting, or is your desire to create a buffer but defer setting a valid address?

timcanham, 2021-07-07:
The use case was creating a `Fw::Buffer` instance for later use. The developer would set all fields to 0 to get it in a known state. That was to address the fact that the old `Fw::Buffer` didn't set the defaults to anything, and would accept 0 as an initializer. I think we can leave it as is, since the new class zeros everything in the constructor.

LeStarch, 2021-07-07:
@timcanham I think that is the right approach.  If you want a NULL or uninitialized buffer, just use the default constructor.

closed 2021-07-07 by LeStarch