nasa/fprime #817: Add a formatter?

AdvaitDhingra, 2021-07-12:
| | |
|:---|:---|
|**_F´ Version_**| n/A|
|**_Affected Component_**|  All of them|
---
## Feature Description

Add a formatter (like clang-format) to keep all of the code formatted in the same way.

## Rationale

A lot of C++ projects (like root-project/root) have this and it is very helpful.

Joshua-Anderson, 2021-07-19:
I'm working on configuring clang-format for F'. It's a bit complicated by the fact that larger project structural changes are happening to integrate FPP into fprime. I believe the current plan format the code after the 3.0 release so that it doesn't cause messy merge conflicts with the FPP changes.

LeStarch, 2022-09-30:
Note: we have a clang-format file, but need to understand how to run it give errors from the `PRIVATE:` construct.  We've discussed removing this or writing a wrapper script.

closed 2023-01-18 by thomas-bc