nasa/fprime #838: fprime-util ignores user requested CMAKE_BUILD_TYPE

saba-ja, 2021-07-16:
| | |
|:---|:---|
|**_F´ Version_**| 2.0 |
|**_Affected Component_**| CMake, fprime-tools  |
---
## Problem Description

## Issue:
Currently it is not possible to export symbols for the sake of debugging when building a deployment (not ut) using fprime-util generate.

## Root cause:
Conflicting logic

The following lines in the fprime-tools builder assign `BUILD_NORMAL` or `BUILD_TESTING` as the build type without checking  the CMAKE_BUILD_TYPE flag provided by the user:

https://github.com/fprime-community/fprime-tools/blob/b8c910283479b7948e4b34c8ed4046da0c4f1d51/src/fprime/util/build_helper.py#L317-L325
```c++
def utility_entry(args):
    """Main interface to F prime utility"""
    parsed, cmake_args, make_args, parser = parse_args(args)
    cwd = Path(parsed.path)

    build_type = BuildType.BUILD_NORMAL
    if parsed.ut:
        build_type = BuildType.BUILD_TESTING
```

Therefore the `CMAKE_BUILD_TYPE` will be always either `Release` or `Testing`.

Since the `CMAKE_BUILD_TYPE` is always set by the fprime-util builder the following CMake lines will ignore user provided build type flag:
https://github.com/nasa/fprime/blob/46e71784484f773682bd32c2d25e482d5c4ff5bb/cmake/Options.cmake#L110-L115

## How to Reproduce

From the Ref deployment folder run:
1. fprime-util generate -DCMAKE_BUILD_TYPE=Testing
or 
2. fprime-util generate -DCMAKE_BUILD_TYPE=Debug

## Expected Behavior

User should be able to export Debug symbols on deployments using fprime-util and `CMAKE_BUILD_TYPE` flag.

Joshua-Anderson, 2021-08-02:
fprime-util doesn't let you manually override the `CMAKE_BUILD_TYPE` variable because it's supposed to abstract that away from the user. Based on the command you're trying to run, it'll automatically infer the correct build cache and build type. I made the mistaken assumption that since we could correctly infer the build type that there wouldn't be any reason for the user to need to override it. 

It looks like you're trying to override the build type because you're looking to enable debug builds. You can do that right now by running the generate step yourself manually `mkdir uild-fprime-automatic-native && cmake .. -DCMAKE_BUILD_TYPE=Debug` or by just setting the `-g` flag in your C++ flags.

Does the Debug build type even properly build right now? The cmake code has a lot of conditional  check based on the build type and I don't think we're accounting for debug anywhere.

Instead of updating `fprime-util` to properly handle when users attempt to manually override the build type, I'd advocate for adding first class support for the debug build type instead.

Then you could do `fprime-util generate --debug` and get a `build-fprime-automatic-native-debug` folder, then run `fprime-util build --debug` to build a debug project.

LeStarch, 2021-12-20:
Fixed in v3.  Further work can be discussed in fprime-tools repo.

closed 2021-12-20 by LeStarch