nasa/fprime #856: Os unit test fails intermittently on Mac OS X

bocchino, 2021-07-20:
| | |
|:---|:---|
|**_F´ Version_**| v3.0.0 |
|**_Affected Component_**| N/A  |
---
## Problem Description

When I run the Os unit tests, on my Mac (MacBook Pro, Mac OS 10.15.1), I get this:
```
1: [ RUN      ] Nominal.IntervalTimerTest
1: /Users/bocchino/JPL/Tools/FPrime/bocchino/fprime/Os/test/ut/IntervalTimerTest.cpp:16: Failure
1: Expected: (timer.getDiffUsec()) < (1005000), actual: 9011207 vs 1005000
1: [  FAILED  ] Nominal.IntervalTimerTest (9011 ms)
```

## How to Reproduce

Run the Os unit tests. The behavior is intermittent; sometimes everything passes.

## Expected Behavior

The tests should pass.

Joshua-Anderson, 2021-07-20:
I've encountered this as well, I'll take a look at it this afternoon.

LeStarch, 2021-07-27:
This was fixed....OS "feature" that prevents this test working correctly.  Disabled on OSX

PR #862 (2021-07-23): Disable interval timer test on Mac OS
| | |
|:---|:---|
|**_Originating Project/Creator_**| |
|**_Affected Component_**|  Os |
|**_Affected Architectures(s)_**|  |
|**_Related Issue(s)_**| fixes #856 |
|**_Has Unit Tests (y/n)_**|  |
|**_Builds Without Errors (y/n)_**|  |
|**_Unit Tests Pass (y/n)_**|  |
|**_Documentation Included (y/n)_**|  |

---
## Change Description

The interval timer unit test is timed off a 1 sec thread delay. Mac OS allows a large amount of scheduling jitter to conserve energy, which rarely causes this sleep to be slightly shorter (~0.99 s) or longer (~10 sec) than req [... cut]
closed 2021-07-27 by LeStarch