nasa/fprime #864: Tutorial.md Possibly Incorrect Install Instructions

astroesteban, 2021-07-22:
| | |
|:---|:---|
|**_F´ Version_**|latest devel |
|**_Affected Component_**| Tutorial.md  |
---
## Problem Description

I am following the instructions in _Tutorial.md_ in an Ubuntu 20.04 Docker container.  In the section **Installing the Ref Deployment and Running the Binary Assuming Linux** of _Tutorial.md_ I followed the steps listed:

```sh
cd fprime/Ref
fprime-util build
./bin/Linux/Ref # Run the deployment
CTRL-C # Exit the application
```

and there is no _.bin_ directory in _fprime/Ref_. 

I ran a `find . -name Ref` and `ls` all over the _fprime_ and _Ref_ directories and I have not found that binary directory. 

## How to Reproduce

1. Setup the latest `fprime/devel` branch on Ubuntu.
2. Go through the _Tutorial.md_ instructions.
3. Build the Ref component and look for the _.bin_ directory.

## Expected Behavior

Be able to walk through _Tutorial.md_ and find the _.bin_ directory on Linux.

rawinza555, 2021-07-23:
The executable file is in build-artifact/Linux/bin if that's what you are looking for.

closed 2021-07-26 by astroesteban

ServiceDog, 2021-08-22:
This is the output of `fprime-util generate` and `fprime-util build`: 
```
build-fprime-automatic-native
├── bin
│   └── Linux
│       └── Reference
├── CMakeCache.txt
├── CMakeFiles
├── cmake_install.cmake
├── F-Prime
├── hashes.txt
├── lib
├── Makefile
└── Ref
```

See the path is `<artifacts>/bin/Linux/<binary>`

Executing `fprime-gds -r build-fprime-automatic-native` I get the error:

```zsh
➜  Reference git:(master) ✗ fprime-gds -r build-fprime-automatic-native
[ERROR] binary location build-fprime-automatic-native/Linux/bin does not exist
```

See the path searched for is '<artifacts>/Linux/bin', not 'bin/Linux'. Is this new or is there something I'm missing?

Joshua-Anderson, 2021-08-23:
@ServiceDog Since F' 2.0, while the binaries are exist in `build-fprime-automatic-native`, the GDS now looks in `build-artifacts` as the primary location for project binaries