nasa/fprime #881: Drv/TcpClient unit tests intermittently fail on Mac OS

bocchino, 2021-08-01:
| | |
|:---|:---|
|**_F´ Version_**| release/v3.0.0|
|**_Affected Component_**| Drv/TcpClient  |
---
## Problem Description

Intermittently, when running the unit tests for Drv/TcpClient, I see errors like this:
```
1: [ RUN      ] Nominal.BasicMessaging
1: /Users/bocchino/JPL/Tools/FPrime/fprime/Drv/TcpClient/test/ut/Tester.cpp:42: Failure
1: Expected equality of these values:
1:   serverStat
1:     Which is: -9
1:   SOCK_SUCCESS
1:     Which is: 0
1: [  FAILED  ] Nominal.BasicMessaging (3 ms)
1: [ RUN      ] Nominal.BasicReceiveThread
1: /Users/bocchino/JPL/Tools/FPrime/fprime/Drv/TcpClient/test/ut/Tester.cpp:42: Failure
1: Expected equality of these values:
1:   serverStat
1:     Which is: -9
1:   SOCK_SUCCESS
1:     Which is: 0
1: [  FAILED  ] Nominal.BasicReceiveThread (1 ms)
```
## How to Reproduce

Run the unit tests as described above. I am running on a MacBook Pro with Mac OS 10.15.17.

## Expected Behavior

Unit tests should pass.

Joshua-Anderson, 2021-08-02:
@LeStarch I seem to remember you said that you found the issue causing this. What was the fix?

LeStarch, 2021-08-02:
This should work and it doesn't....so apparently I did not fix it.

LeStarch, 2021-08-02:
...wait this is on the v3.0.0 branch.  We should update to the latest devel and see if it fails there.

LeStarch, 2021-12-15:
Not seen this in a few months on v3.0.0 w/ macOS CI.  Closing.

closed 2021-12-15 by LeStarch

JohanBertrand, 2024-04-22:
Out of curiosity, what is the reason behind this error?

It is also randomly happening for me on a docker

```
[ RUN      ] Nominal.BasicMessaging
/builds/Fprime/Drv/TcpServer/test/ut/TcpServerTester.cpp:44: Failure
Expected equality of these values:
  serverStat
    Which is: -9
  SOCK_SUCCESS
    Which is: 0
/builds/Fprime/Drv/TcpServer/test/ut/TcpServerTester.cpp:46: Failure
Value of: component.getSocketHandler().isStarted()
  Actual: false
Expected: true
/builds/build-fprime-automatic-native-ut/F-Prime/Drv/TcpServer/TcpServerGTestBase.cpp:87: Failure
Expected equality of these values:
  size
    Which is: 1
  this->fromPortHistorySize_ready
    Which is: 0
/builds/Fprime/Drv/TcpServer/test/ut/TcpServerTester.cpp:100
  Value:    Size of history for ready
  Expected: 1
  Actual:   0
[  FAILED  ] Nominal.BasicMessaging (0 ms)
```

JohanBertrand, 2024-04-25:
For the record, it is still happening on the pipeline, but not only on macos: https://github.com/nasa/fprime/actions/runs/8839296509/job/24272334136

```
[ RUN      ] Reconnect.MultiMessaging
/home/runner/work/fprime/fprime/Drv/TcpServer/test/ut/TcpServerTester.cpp:44: Failure
Expected equality of these values:
  serverStat
    Which is: -9
  SOCK_SUCCESS
    Which is: 0
/home/runner/work/fprime/fprime/Drv/TcpServer/test/ut/TcpServerTester.cpp:46: Failure
Value of: component.getSocketHandler().isStarted()
  Actual: false
Expected: true
/home/runner/work/fprime/fprime/build-fprime-automatic-native-ut/F-Prime/Drv/TcpServer/TcpServerGTestBase.cpp:87: Failure
Expected equality of these values:
  size
    Which is: 100
  this->fromPortHistorySize_ready
    Which is: 0

/home/runner/work/fprime/fprime/Drv/TcpServer/test/ut/TcpServerTester.cpp:100
  Value:    Size of history for ready
  Expected: 100
  Actual:   0

[  FAILED  ] Reconnect.MultiMessaging (1 ms)
```