nasa/fprime #892: Throttle send frame EVR

timcanham, 2021-08-04:
| | |
|:---|:---|
|**_F´ Version_**|commit d3fa31c |
|**_Affected Component_**| ?  |
---
## Problem Description

A description of the problem with sufficient detail to understand the issue.

If there is no ground system, the interface continuously sends this pair of EVRs:

```
0x201db690 (TV_TLM): [ERROR] Failed to send framed data: 0
0x201db690 (TV_TLM): [ERROR] Failed to send framed data: 0
0x201db690 (TV_TLM): [ERROR] Failed to send framed data: 0
0x201db690 (TV_TLM): [ERROR] Failed to send framed data: 0
0x201db690 (TV_TLM): [ERROR] Failed to send framed data: 0
0x201db690 (TV_TLM): [ERROR] Failed to send framed data: 0
0x202236f0 (TV_ReceiveTask): [WARNING] Failed to open port with status 61 and errno 0
```

## How to Reproduce

1. Run Ref without the ground system
2.
3.

## Expected Behavior

IMHO the EVR should throttle, and perhaps the throttle is reset when the connection is make.

LeStarch, 2021-08-04:
@timcanham this is likely a log message produced in the framer component...or IP driver.  Since this is a failure to send, it isn't an EVR because it is indicating a condition where sending EVRs isn't working.

timcanham, 2021-08-04:
Good point... It would be nice to figure out a way to throttle it.

Joshua-Anderson, 2021-08-04:
This is also a poor first time user experience, since they'll always see errors while waiting for the GDS to connect. Maybe we can add some logic to only print this warning after the initial connection?

LeStarch, 2021-12-20:
This isn't exactly accurate as it is not an EVR, but a log message.  I have linked the issue which can we used to fix this.  Closing as a duplicate.

closed 2021-12-20 by LeStarch