nasa/fprime #928: Running the RPI ground station Gds/bin/run_deployment.sh not found

markospacetech, 2021-08-17:
| | |
|:---|:---|
|**2.0**| |
|**RPI**|   |
---
## Problem Description

Follow the RPI instructions https://github.com/nasa/fprime/blob/devel/RPI/README.md ,  after building the project and trying to run the scripts run_rpi_cross.sh, I found that the Gds/bin folder is no where to be found in the project or fprime directory. Could this be an issue with how I installed gds? I followed the latest installation instructions.

## How to Reproduce

1. cd fprime/RPI
    fprime-util generate
2.cd fprime/RPI
   fprime-util build
3. ./scripts/run_rpi_cross.sh

## Expected Behavior
Run the ground station fprime/RPI to be used to connect to raspberrypi

Joshua-Anderson, 2021-08-18:
It looks like that part of the raspberry pi example tutorial got outdated

To cross compile you can just run:

```
fprime-util generate
fprime-util build
```

(the settings.ini file sets up the RPI example to cross compile to the raspberry pi by default)

To start the GDS just run `fprime-gds -n` in the RPI example directory.

markospacetech, 2021-08-18:
@Joshua-Anderson Thanks that worked!

Now I'm running into an issue when trying to run the binary on the pi, 

on this step from the guide 
`./RPI -a <ground computer's IP> -p 50000`
returns:
Assert file "~/fprime/Drv/LinuxGpioDriver/LinuxGpioDriverComponentImpl.cpp": Line: 276 
Assert file "~/fprime/RPI/build-fprime-automatic-raspberrypi/RPI/RpiDemo/RpiDemoComponentAc.cpp": Line: 291
Assert file "~/fprime/RPI/build-fprime-automatic-raspberrypi/RPI/RpiDemo/RpiDemoComponentAc.cpp": Line: 291
**Segmentation fault**

Any help or guidance is much appreciated.

Joshua-Anderson, 2021-08-18:
I'm going to re-open this to track fixing/updating the raspberry pi tutorial so we don't forget about it :)

Emil808, 2021-08-23:
I came here with this same problem, followed @Joshua-Anderson's suggestions and am able to generate, build, and copy the file to the RPI. 
I found other discrepancies within the tutorial. 
I am also encountering `[ERROR] dictionary location <path_to_fprime>/RPI/build-artifacts/Linux/dict does not exist` when trying to start the ground station from the `RPI/` directory. The only directory in `RPI/build-artifacts/` is `raspberrypi/`.

markospacetech, 2021-08-24:
Yeah same here, I just made renamed the folder to /Linux and it worked.

Emil808, 2021-08-24:
Neat, ground station runs now. But now on the raspberry pi, running `./RPI -a <ground IP> -p 50000` I get a segmentation fault.

timcanham, 2021-08-24:
Did you run the app as sudo? It's usually that, or the I/O isn't configured correctly on the board.

Emil808, 2021-08-24:
> Did you run the app as sudo? It's usually that, or the I/O isn't configured correctly on the board.

Yes, I had to run the app with sudo. 
So the weird thing is that `fprime-util build` will result in a `build-artifacts/raspberrypi/` but `fprime-gds -n` is trying to find `build-artifacts/Linux/`

LeStarch, 2021-12-20:
We should update the README to have good instructions.

closed 2021-12-21 by LeStarch