nasa/fprime #930: Out of source build - cmake autocodegenerator fails due to duplicate cmake targets

DentOpt, 2021-08-18:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

I am trying to build an f' application out of source with an infrastructure where all proprietary code modules are in the root. 
However, I get errors of duplicate cmake targets from the fprime cmake stack. 
We assume that the definition of the install directories for code modules does not allow out of source architecture builds with this topology. Do you have any suggestions?
Test is the application that contains the topology: 

```
build
root_repo  
├──Fprime  
│   ├── Autocoders
│   └── ...
└── Test  
│   ├── CMakeLists.txt  
│   ├── settings.ini  
│   └── Topology  
│       ├── CMakeLists.txt  
│       └── TestTopologyAppAi.xml  
│
    .... in the future ...
│
├── Module1 
├── Module2
...
```

### settings.ini  
```
[fprime]  
project_root: ./..  
framework_path: ./../Fprime  
library_locations: ./..  
default_toolchain: native  
environment_file:  
ac_constants:  
config_directory:  
```

### CMakeLists.txt (in Test folder)
```
project(Test C CXX)
cmake_minimum_required(VERSION 3.13)

set(FPRIME_PROJECT_ROOT   ${CMAKE_CURRENT_SOURCE_DIR}/..)
set(FPRIME_FRAMEWORK_PATH ${CMAKE_CURRENT_SOURCE_DIR}/../Fprime)
include("${FPRIME_FRAMEWORK_PATH}/cmake/FPrime.cmake")
include("${FPRIME_FRAMEWORK_PATH}/cmake/FPrime-Code.cmake")

add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/Topology/")
add_fprime_subdirectory("${FPRIME_PROJECT_ROOT}/Fprime/")
```
### CMakeLists.txt (in Topology)
```
set(SOURCE_FILES
  "${CMAKE_CURRENT_LIST_DIR}/TestTopologyAppAi.xml"
)
set(MOD_DEPS
  Fw/Logger
  Svc/LinuxTime
)
register_fprime_executable()
```

## Expected Behavior
[... 185 lines cut ...]
  is already used to build a source directory.  It cannot be used to build
  source directory

    /home/root_repo/Fprime/CFDP

  Specify a unique binary directory name.
Call Stack (most recent call first):
  /home/root_repo/Fprime/CMakeLists.txt:29 (include)

CMake Error at /home/root_repo/Fprime/cmake/FPrime-Code.cmake:32 (add_subdirectory):
  The binary directory

    /home/build/F-Prime/Utils

  is already used to build a source directory.  It cannot be used to build
  source directory

    /home/root_repo/Fprime/Utils

  Specify a unique binary directory name.
Call Stack (most recent call first):
  /home/root_repo/Fprime/CMakeLists.txt:29 (include)

-- Configuring incomplete, errors occurred!
```

DentOpt, 2021-08-18:
We found the issue: 
As in the settings the framework path is defined as 
framework_path: ./../Fprime

`add_fprime_subdirectory("${FPRIME_PROJECT_ROOT}/Fprime/")` is already done internally and therefore should not be in the Test/CmakeLists.txt

closed 2021-08-18 by DentOpt

Joshua-Anderson, 2021-08-18:
@DentOpt actually, I suspect the problem is the `library_locations` attribute. That's meant to allow you to include F' libraries or other top level directories into F' (See https://github.com/fprime-community/fprime-sphinx). Because you set the framework path and a library location to the same location I suspect you added the ./Fprime directory to the search path twice, giving you the error you saw above. By removing the `framework_path` attribute you removed the duplicate entry and were likely including the ./Fprime through `library_locations`. While this works, it may result in some strange behavior. I'd recommend setting `framework_path`.