nasa/fprime #934: Autogenerate separate `Tester::connectPorts` source file

timcanham, 2021-08-18:
| | |
|:---|:---|
|**_F´ Version_**|2.0 |
|**_Affected Component_**|  |
---
## Feature Description

Currently, `Tester::connectPorts()` is generated by the `generate impl --ut` and placed in the `Tester.cpp` file. This becomes a maintenance annoyance and causes asserts if it's not corrected if a new port is added.

The proposal would be to generate this function into a separate .cpp file that would be treated like the other unit test files and generated each time the test is compiled. 

## Rationale

It makes maintaining unit tests a lot easier, and it isn't intuitive to know you have to go update this function when you add a port.

LeStarch, 2021-08-18:
If we are going to fix this, we need to make sure the `for` connections for port arrays use non-hardcoded values.  I often have to rewrite these methods by hand to fix that oversight.

timcanham, 2021-08-18:
What do you mean by `for` connections?

LeStarch, 2021-08-18:
When you have an array port, it makes  a for loop:

```
for (i = 0; i < 3; i++) {
   //connect port @ i
}
```

However, instead of using the constant depicting the array size, it uses a hardcoded integer.  If one changes AcConstants.ini or that port size, this hardcoded int needs to be updated to.

timcanham, 2021-08-18:
If the ut code is regenerated, does the port size get updated?

LeStarch, 2021-08-18:
@timcanham would we be able to override this autocode?  Occasionally we need to override it.

e.g. for the hub unit tests, we create 2 hubs and cross-wire them together to proved that the outside interfaces of the hubs work in tandem.

timcanham, 2021-08-18:
Since the connection functions are always re-callable, you can re-connect any way you like. You could also just comment out the call to `connectPorts()` from the `Tester` constructor.

PR #1786 (2022-12-21): auto-generate connect ports option. Fixes #934
| | |
|:---|:---|
|**_Originating Project/Creator_**| |
|**_Affected Component_**|  |
|**_Affected Architectures(s)_**|  |
|**_Related Issue(s)_**| Fixes #934 |
|**_Has Unit Tests (y/n)_**| y  |
|**_Builds Without Errors (y/n)_**| y |
|**_Unit Tests Pass (y/n)_**| y  |
|**_Documentation Included (y/n)_**| y |

---
## Change Description

Auto-generates `connectComponents` and `initComponents` helper functions into a `TesterHelpers.cpp` file. To use, a user sets the variable `set(UT_AUTO_HELPERS ON)`.

It also moves the UT `#define` settings to `Tes [... cut]
commit e596a2bc (2022-12-21): auto-generate connect ports option. Fixes #934 (#1786)

* auto-generate connect ports option. Fixes #934

* updating math tutorial

* making unit tests settings from #defines to constants

* fixing missing properties in signal gen
closed 2022-12-21 by github-actions[bot]