nasa/fprime #950: Improvement of NMEA message parsing on GPS tutorial

rawinza555, 2021-08-22:
| | |
|:---|:---|
|**_F´ Version_**| 2.0.0|
|**_Affected Component_**| gpscomponentimpl.cpp |
---
## Feature Description

Minor improvement can be added for nmea parsing message to include lat long data from GLONASS or other satellite. In other word, NMEA parsing part of the component's code should include message starting with "$GLGGA" and "$GNGGA" as well. 

Also I'm not sure if this is more of a UART issues but my gps module only send one char at a time for NMEA message when connected directly using UART TX and RX. Perhaps the module you have that use USB doesn't have this kind of issue.  

## Rationale
I imagined this is probably going to be in a very low priority but Include this will ensure more data is received and processed.

Joshua-Anderson, 2021-08-25:
One thing that needs to be balanced with the GPS app is keeping it simple enough that it makes a good tutorial for new users to follow vs continuing to make it more functional by adding support for parsing more NMEA codes. The current parser doesn't handle `GPGGA` messages correctly when the GPS unit doesn't have a lock either.  This has caused confusion in users following the tutorial and thinking that something was broken rather than their unit just didn't have a lock.

That being said, if we can parse GLGGA and GNGGA messages without adding a lot of complexity to the tutorial I'd be all for it.

LeStarch, 2021-12-20:
I am going to close this.  The GPS tutorial is, as @Joshua-Anderson  said, supposed to be a simple example.  Real users can implement their own component that is better.

closed 2021-12-20 by LeStarch