nasa/fprime #1044: Math Tutorial section 2.4.1.3.2: "Undefined symbols for architecture x86_64"

rpg0011, 2021-09-17:
I'm running into an issue with the Math tutorial, section 2.4.1.3.2
After I run the "fprime-util generate --ut" command and then run the "fprime-util build --ut" inside of the MathSender folder
I get the error below. I'm wondering if it has something to do with me using an M1 processor?

                Undefined symbols for architecture x86_64:
                  "Ref::Tester::testDivCommand()", referenced from:
                      Nominal_DivideOperationTest_Test::TestBody() in TestMain.o
                  "Ref::Tester::testSubCommand()", referenced from:
                      Nominal_SubOperationTest_Test::TestBody() in TestMain.o
                  "Ref::Tester::testMultCommand()", referenced from:
                      Nominal_MultOperationTest_Test::TestBody() in TestMain.o
                ld: symbol(s) not found for architecture x86_64
                clang: error: linker command failed with exit code 1 (use -v to see invocation)
                make[3]: *** [bin/Darwin/Ref_MathSender_ut_exe] Error 1
                make[2]: *** [Ref/MathSender/CMakeFiles/Ref_MathSender_ut_exe.dir/all] Error 2
                make[1]: *** [Ref/MathSender/CMakeFiles/Ref_MathSender_ut_exe.dir/rule] Error 2
                make: *** [Ref_MathSender_ut_exe] Error 2
                [ERROR] CMake erred with return code 2

I've also tried running in the test/ut folder and get a different error listed below:

                Policy CMP0048 is not set: project() command manages VERSION variables.
                Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
                command to set the policy and suppress this warning.
              
                The following variable(s) would be set to empty:
              
                  CMAKE_PROJECT_VERSION
                  CMAKE_PROJECT_VERSION_MAJOR
                  CMAKE_PROJECT_VERSION_MINOR
                  CMAKE_PROJECT_VERSION_PATCH
              This warning is for project developers.  Use -Wno-dev to suppress it.
              
              make: *** No rule to make target `Ref_MathSender_test_ut_ut_exe'.  Stop.
              [ERROR] CMake erred with return code 2

LeStarch, 2021-09-21:
The second error is a CMake warning.  Newer versions of CMake want versioning information placed in the `project()` call.  I'd ignore that second warning for now.

LeStarch, 2021-09-21:
The primary error has to do with the sollowing symbols:

```
              "Ref::Tester::testDivCommand()"
              "Ref::Tester::testSubCommand()"
              "Ref::Tester::testMultCommand()"
```

Did you add those three functions to the Tester.cpp?  Is the Tester.cpp for the math component added to the CMake source list in CMakeLists.txt?

LeStarch, 2021-12-20:
Didn't see a follow-up.  Closing.

closed 2021-12-20 by LeStarch