nasa/fprime #1057: Add history check for value of enum serializable class rather than enum class instance

timcanham, 2021-09-23:
| | |
|:---|:---|
|**_F´ Version_**| 2.0 |
|**_Affected Component_**| All |
---
## Feature Description

When unit testing and checking for the receipt of an external enumeration class in the history of telemetry or events, the user is required to create an instance of the class, assign a value to it, and then pass the instance to the checking function.

Proposal would be to add an overloaded version of the checker such that the enumeration value can be checked.

## Rationale

Makes unit test code less cluttered.

matt392code, 2025-01-03:
[fprime-enum-test.txt](https://github.com/user-attachments/files/18304200/fprime-enum-test.txt)
Proposed solution attached.
The benefits of this enhancement would be:
1. More concise test code:
   - Eliminates boilerplate enum instance creation
   - Reduces lines of code needed for checks
   - Makes tests more readable
2. More direct value checking:
   - Tests can directly reference enum values
   - Reduces potential for errors in enum instance setup
   - Makes the intent of the test clearer
3. Better maintainability:
   - Less code to maintain
   - Easier to update when enum values change
   - More straightforward test logic
The proposal aligns well with F' goals of improving developer experience while maintaining robust testing capabilities.

LeStarch, 2025-02-04:
Is this still an issue, or can I close?

timcanham, 2025-02-04:
I hadn't heard that any change addressed it.