nasa/fprime #1060: RPI Tutorial: Failed to uplink and open socket

juliannerice, 2021-09-24:
| | |
|:---|:---|
|**_2.0.0_**| |
|**_RPI Tutorial_**|   |
---
## Problem Description

After following the RPI Demo Tutorial, when trying to connect between the GUI and RPI every command uplink fails. Exiting the 
GUI results in a message that the socket at 0.0.0.0:50000 failed to open. 
[image]

## How to Reproduce

1. Follow RPI Tutorial
2. Use command _fprime-gds -n_ instead of _./scripts/run_rpi_cross.sh_ to launch GUI
3. Send a command in the GUI
4. Exit GUI in the terminal

## Expected Behavior
After sending a command in the GUI, the uplink should fail to send the respective number of bytes of data after 3 retries. When exiting the GUI, an error should appear: _Failed to open socket at 0.0.0.0:50000, retrying: OSError: [Errno 22] Invalid argument_

timcanham, 2021-09-24:
@juliannerice Have you check firewall settings? The ports might be blocked.

juliannerice, 2021-09-27:
Hi there! I have checked the firewall settings and none of the respective ports should be blocked. It's weird because I got it to work a couple times, but it hasn't been consistent.

LeStarch, 2021-12-20:
We haven't been able to reproduce, but this kind of inconsistency usually means firewall or router settings.

closed 2021-12-20 by LeStarch