nasa/fprime #1072: Utilize and Statically Analyze for JPL Coding Guidelines

Joshua-Anderson, 2021-09-30:
## Feature Description

[JPL's Institutional Coding Standards for C](https://yurichev.com/mirrors/C/JPL_Coding_Standard_C.pdf) is an institution wide standard for C code. While JPL has no C++ coding standard, many of the C coding standard requirements (no memory allocation outside of initialization, no recursion, etc) can be adapted to C++ projects like F'.

Requirements from the JPL C coding standard requirements that apply to C++ projects should be selected and F' should be audited for requirement compliance.

We should consider using static analysis to audit for and possible enforce compliance with the JPL C coding guidelines.

GitHub has open source CodeQL scripts for most JPL coding standard and power of ten rules available [here](https://github.com/github/codeql/tree/main/cpp/ql/src/). We could use these rules with github actions code scanning or custom LGTM analysis scripts. My first glance over the CodeQL scripts revealed that many of them are VxWorks specific and may require refactoring to work with F'

ThibFrgsGmz, 2021-10-01:
Very good idea! Particularly since some of your rules make reference to MISRA-C. Some of our quality engineers enjoy "bugging" us to follow MISRA-C (2004) guidelines.

This would reassure the various stakeholders' quality engineers, who may be hesitant if internationally recognized coding rules are not followed.

ThibFrgsGmz, 2021-11-19:
I discovered C++ rules on the [NASA Technical Reports Server](https://ntrs.nasa.gov/citations/20080039927).
You don't consider them because they come from another center, are in draft status, and are out of date?

Is the JPL working on writing coding rules for the C++ language?

If so, do you use the [MISRA C++ rules](https://www.misra.org.uk/misra-c-plus-plus/), the [ESA C++ rules](https://vdocument.in/c-and-c-coding-standards-military-standards-mil-c-and-c-coding-standards-.html), or the [Lockheed Martin JSR AV C++ rules](https://www.stroustrup.com/JSF-AV-rules.pdf)?

LeStarch, 2022-11-29:
This is done!

closed 2022-11-29 by LeStarch