nasa/fprime #1074: Fw Buffer Issue Fatal 141

DentOpt, 2021-10-01:
| | |
|:---|:---|
|**_F´ Version NASA_v1.5.3**| |
|**_Affected Component_**|   |
---
# Problem Description

Application fails when starting the event logger in `Topology.cpp`
`eventLogger.start(0,98,10*1024);`

```
   Svc::ActiveLoggerImpl eventLogger(FW_OPTIONAL_NAME("ELOG"));
...
   eventLogger.init(10,0); 
... 
    // Active component startup
    // start rate groups
    rateGroup1.start(0, 121,10 * 1024);
    rateGroup2.start(1, 120,10 * 1024);
    rateGroup3.start(2, 119,10 * 1024);
    rateGroup4.start(3, 118,10 * 1024);
    // start driver
    blockDrv.start(0,140,10*1024);
    // start dispatcher
    cmdDisp.start(0,101,10*1024);
    // start sequencer
    cmdSeq.start(0,100,10*1024);
    // start telemetry
    eventLogger.start(0,98,10*1024);      //<==================Fails here
    chanTlm.start(0,97,10*1024);
    prmDb.start(0,96,10*1024);
    fileDownlink.start(0, 100, 10*1024);
    fileUplink.start(0, 100, 10*1024);
    fileManager.start(0, 100, 10*1024);
    pingRcvr.start(0, 100, 10*1024);
...
```
If "eventLogger.start(0,98,10*1024)" is commented out, the same error appears at pingRcvr.start(0, 100, 10*1024);.

## Error message
`build-fprime-automatic-native/F-Prime/Fw/Buffer/FATAL 141 handled.`
Do you have any idea what this buffer error is referring to?

# Problem trace
Fprime/Os/Posix/Task.cpp failing in:
stat = pthread_create(tid,&att,(pthread_func_ptr)routine,arg);

Variables:
[image]
___

Called in Fprime/Fw/Comp/ActiveComponentBase.cpp by: 
Os::Task::TaskStatus status = this->m_task.start(taskName, identifier, priority, stackSize, this->s_baseTask, this, cpuAffinity);

`task_name = "ELOG\0"` as expected
Variables:
[image]

___

## Buffer configuration
Fprime/config/FpConfig.hpp:
```
/**
 * \file
 * \author T. Canham
 * \brief ISF configuration file
 *
 * \copyright
 * Copyright 2009-2015, by the California Institute of Technology.
[... 322 lines cut ...]
#ifndef FwTimeBaseStoreType
#define FwTimeBaseStoreType U16                 //!< Storage conversion for time base in scripts/ground interface
#endif

#ifndef FwTimeContextStoreType
#define FwTimeContextStoreType U8                 //!< Storage conversion for time context in scripts/ground interface
#define FW_CONTEXT_DONT_CARE 0xFF                 //!< Don't care value for time contexts in sequences
#endif

// These setting configure whether or not the timebase and context values for the Fw::Time
// class are used. Some systems may not use or need those fields

#ifndef FW_USE_TIME_BASE
#define FW_USE_TIME_BASE                1 //!< Whether or not to use the time base
#endif

#ifndef FW_USE_TIME_CONTEXT
#define FW_USE_TIME_CONTEXT             1 //!< Whether or not to serialize the time context
#endif

// *** NOTE configuration checks are in Fw/Cfg/ConfigCheck.cpp in order to have
// the type definitions in Fw/Types/BasicTypes available.

#endif
```

LeStarch, 2021-10-01:
Depending on the Os you are running on, the stack size could be bad.  You are passing in 10KB, but the linux minimum is 16KB.   However, these values are ignored in 1.5.3 running `Os/Posix/Task.cpp`  as it is designed for user space where you do not have permissions to set these items.

This behavior has changed in devel: https://github.com/nasa/fprime/discussions/1041

LeStarch, 2021-10-01:
Hmmm....doesn't seem to be the case, as the first few starts would likely fail first.

DentOpt, 2021-10-03:
@LeStarch 
Thank you for your input. I tried with 16KB and 32KB unfortunately without a result.

LeStarch, 2021-12-20:
Closing, as not followed up.  Please feel free to follow-up or ask in discussions.

closed 2021-12-20 by LeStarch