nasa/fprime #1108: Possible improvements to Math tutorial

timmyd7777, 2021-11-15:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

One of the last sections of the Math Component tutorial (4.1.8.1 Setting the Parameter Value) is missing some examples or possibly graphics.  These sentences are incomplete:

- The FACTOR1_PRM_SET command can be sent to the software:
- The notification function that was implemented as part of the tutorial will send an event indicating the value was updated:
- The MS_DO_MATH command can now be executed with the new value:
- The MathReceiver component sends the events with the new result:
- The new result is 1.0 with the new value of factor2. The "Channel Telemetry" tab also shows the new values:

I figured it out, but you may want to fix this.

Also, in section 4.1.8.2 Saving the Parameter Value, we are saving the value of FACTOR2, not FACTOR1.  But the corresponding variable (factor2) does not get used anywhere in the generated C++ code.  I would think that saving factor1 to the parameter database makes more sense.  If that's left as an exercise for the reader, then you might want to state so.

## Rationale

The tutorial will make more sense to new users with these changes.

LeStarch, 2021-12-20:
We need to verify this against the new math tutorial and ensure these are fixed if still present.

LeStarch, 2022-05-03:
I just looked through these and I don't see these partial sentences in the math component tutorial rewrite w/ FPP.  Thus, I am closing this issue.  If the issues persist in the rewrite, let us know.

closed 2022-05-03 by LeStarch