nasa/fprime #1137: Array Creation Uns

DentOpt, 2021-12-04:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

A description of the problem with sufficient detail to understand the issue.

## How to Reproduce

[/Types/Vector3DSerializableAi.xml](/Types/Vector3DSerializableAi.xml)

```
<serializable namespace="Types" name="Vector3D">
    <comment>
    Vector class for description of 3D vectors
    </comment>
    <members>
        <member name="vec" type="F32" size="3"/>
    </members>
</serializable>
```
generates the interface which is unsafe. 

[build-fprime-automatic-native/Types/Vector3DSerializableAc.cpp](build-fprime-automatic-native/Types/Vector3DSerializableAc.cpp)

```
const F32* Vector3D::getvec(NATIVE_INT_TYPE& size) {
    size = 3;
    return this->m_vec;
}
```

## Expected Behavior

A description of the expected behavior.

LeStarch, 2022-11-30:
@bocchino was this fixed by the new array autocoding?  If so, would you close this?

bocchino, 2022-11-30:
Yes, this is fixed:
* As of FPP v1.1.0, we have FPP-to-C++ code generation for structs a.k.a. SerializableAc.cpp files
* In the FPP-generated code the getter method for a member array returns a const or non-const reference to the array type. The array size is encoded in the return type. No raw pointers are returned.

closed 2022-11-30 by bocchino