nasa/fprime #1145: Make the stub Linux drivers emulate nominal behavior

bocchino, 2021-12-09:
As I revise the RPI demo, I'm encountering the following behavior: if I put in flight-like assertions that driver initialization is OK, then the app crashes on the Mac. I presume this is because the stub drivers return error status on the Mac, even when nothing is wrong. Presumably it would work better on the RPI, but we want the demo to behave nicely on Mac/Linux. So for now I am putting in a bunch of code to handle error cases and behave nicely if things fail. This code is non-flight-like, and somewhat annoying to write.

I think it would be better if we had stub drivers always do something reasonable and always return OK. That way we could plug a stub driver into flight-like code that says FW_ASSERT(driverInit()) and everything will still work. The driver behavior can still be very limited for the demo, but at least it won't communicate a failure back that the application layer has to handle in a non-flight-like way.

LeStarch, 2021-12-20:
We should add in serial stubbed driver as requested by another issue.

LeStarch, 2021-12-20:
Another clean-up issue: #1066

matt392code, 2025-01-02:
[stub-driver.txt](https://github.com/user-attachments/files/18294394/stub-driver.txt)
Proposed stub driver implementation . Here's the key approach taken:
1. All operations (open, close, read, write) return DRIVER_OK by default
2. The driver maintains basic state tracking (initialized flag)
3. Read operations return dummy data to simulate real hardware
4. Write operations log the action but always succeed
5. All flight-like assertions are maintained
This implementation solves several problems:
- Applications can use flight-like assertions without crashing
- The code works consistently across Mac, Linux, and RPI platforms
- Development and testing can proceed without hardware dependencies
- The interface remains identical to flight code

To implement this for the serial driver mentioned by LeStarch, you would follow the same pattern but customize the dummy data and logging to match serial port behavior.

closed 2025-04-11 by thomas-bc