nasa/fprime #1163: Cross-compile toolchain for raspberry pi is deprecated

capsulecorplab, 2021-12-15:
[Cross-compile toolchain](https://github.com/raspberrypi/tools) in [package installation](https://github.com/nasa/fprime/blob/devel/RPI/README.md#package-installation) for raspberry pi demo is listed as a deprecated project.

LeStarch, 2021-12-16:
This is true.  Installing the `gcc-arm-linux-gnueabihf` toolchain, and setting the RPI_TOOLCHAIN_DIR environment variable to the install directory should allow using the newer toolchain.

i.e. ${RPI_TOOLCHAIN_DIR}/bin/ should contain the binary tools

capsulecorplab, 2021-12-16:
Installing `gcc-arm-linux-gnueabihf` with `apt` on Ubuntu puts the binaries in `/usr/bin`, so I set `RPI_TOOLCHAIN_DIR` to `/usr/bin`. After doing so, I also had to install `g++-arm-linux-gnueabihf` to resolve a CMake error when running `fprime-util generate`. I'm wondering if it'd be more useful if the installation instructions for the cross-compile toolchain were more explicit in the [package installation](https://github.com/nasa/fprime/blob/devel/RPI/README.md#package-installation) section, so that readers following the deprecated link don't have to debug these issues.

capsulecorplab, 2021-12-16:
Installing `gcc-arm-linux-gnueabihf` with `apt` on Ubuntu puts the binaries in `/usr/bin`, so I set `RPI_TOOLCHAIN_DIR` to `/usr`. After doing so, I also had to install `g++-arm-linux-gnueabihf` to resolve a CMake error when running `fprime-util generate`. I'm wondering if it'd be more useful if the installation instructions for the cross-compile toolchain were more explicit in the [package installation](https://github.com/nasa/fprime/blob/devel/RPI/README.md#package-installation) section, so that readers following the deprecated link don't have to debug these issues.

capsulecorplab, 2021-12-21:
Is there a way to run the RPI CI build locally? Working on a PR to close this issue

LeStarch, 2021-12-21:
I have yet to find a good way of dealing with Github Actions locally.  I just push to github and see how it works out.  Then adjust if it fails.

capsulecorplab, 2021-12-21:
I updated `build-test-rpi.yml` and `RPI.bash` to reflect the recommended installation steps for `gcc-arm-linux-gnueabihf` and `g++-arm-linux-gnueabihf` and pushed to github to trigger the CI, but I'm getting this error I don't know how to debug https://github.com/capsulecorplab/fprime/runs/4590267912?check_suite_focus=true

LeStarch, 2021-12-21:
You can find the full logs by:

1. Clicking on "Summary" in the Actions' Left Pane
2. Scroll to the bottom to find the artifacts section
3. Downloading the rpi-logs artifact

The error is coming from the generate step:

```
CMake Error at /home/runner/work/fprime/fprime/cmake/toolchain/raspberrypi.cmake:26 (message):
  RPI toolchain not found at
  /opt/rpi/tools/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf.  Install it, set
  RPI_TOOLCHAIN_DIR environment variable, or choose another toolchain
Call Stack (most recent call first):
  /usr/local/share/cmake-3.22/Modules/CMakeDetermineSystem.cmake:124 (include)
  CMakeLists.txt:21 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
[ERROR] CMake erred with return code 1. Partial build cache remains. Run purge to clean-up.
```

LeStarch, 2021-12-21:
If you are changing the compiler, you'll likely need to update the top of this file where it infers the toolchain path when not set as an environment variable:

https://github.com/nasa/fprime/blob/devel/cmake/toolchain/raspberrypi.cmake

LeStarch, 2021-12-21:
One other note:  once the first stage (build and compile) passes, submit a PR to the fprime repo.  The second stage action cannot run outside of `nasa/fprime`.

LeStarch, 2021-12-21:
Also assigned you since you are working on this.  If it becomes too much, we can reassign it.

capsulecorplab, 2021-12-22:
I tried setting the `RPI_TOOLCHAIN_DIR` environment variable in [raspberrypi.cmake](https://github.com/capsulecorplab/fprime/commit/c16cfac176a1fefa66b7afda1025ba5bf99c9fac), but I'm still getting a CMake error that it can't find the RPI toolchain. 

Error from [rpi-logs artifact](https://github.com/capsulecorplab/fprime/actions/runs/1613420721):

```
CMake Error at /home/runner/work/fprime/fprime/cmake/toolchain/raspberrypi.cmake:27 (message):
  RPI toolchain not found at
  /opt/rpi/tools/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf.  Install it, set
  RPI_TOOLCHAIN_DIR environment variable, or choose another toolchain
Call Stack (most recent call first):
  /usr/local/share/cmake-3.22/Modules/CMakeDetermineSystem.cmake:124 (include)
  CMakeLists.txt:21 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
[ERROR] CMake erred with return code 1. Partial build cache remains. Run purge to clean-up.
```

astroesteban, 2022-01-07:
A while back I made a cross-compilation toolchain for Raspberry Pi using that new compiler toolchain you mentioned.

I installed the toolchain with:
```sh
apt-get install -y gcc-arm-linux-gnueabihf g++-arm-linux-gnueabihf
```
on Ubuntu 20.04

 The CMake toolchain looks like this:

```cmake
####
# Raspberry PI Toolchain
#
# This is a toolchain for the Raspberry Pi. This toolchain can be used ot build
# against the Raspberry Pi embedded Linux target. In order to use this toolchain,
# the Raspbery Pi cross-compiler should be installed on a Linux host. These
# tools are installable as follows:
#   sudo apt install gcc-arm-linux-gnueabihf g++-arm-linux-gnueabihf
####

# Set the system information
set(CMAKE_SYSTEM_NAME       Linux)
set(CMAKE_SYSTEM_PROCESSOR  arm)
set(CMAKE_SYSTEM_VERSION    1)

# Set the GNU ARM toolchain
set(CMAKE_AR            arm-linux-gnueabihf-ar)
set(CMAKE_C_COMPILER    arm-linux-gnueabihf-gcc)
set(CMAKE_CXX_COMPILER  arm-linux-gnueabihf-g++)
set(CMAKE_ASM_COMPILER  arm-linux-gnueabihf-as)
set(CMAKE_OBJCOPY       arm-linux-gnueabihf-objcopy)
set(CMAKE_OBJDUMP       arm-linux-gnueabihf-objdump)
set(CMAKE_C_GDB         gdb-multiarch)

# Configure the find commands for finding the toolchain
set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM   NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY   NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE   NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE   NEVER)
```

I didn't need to provide the full path to the compiler binaries since they were installed in the system-wide default locations so CMake automagically found them.

Maybe something like this could help your cause?

capsulecorplab, 2022-05-08:
Apologies for the delay, but replacing the [raspberrypi.cmake](https://github.com/nasa/fprime/blob/devel/cmake/toolchain/raspberrypi.cmake) with [@astroesteban's raspberrypi.cmake](https://github.com/nasa/fprime/issues/1163#issuecomment-1007810914) seems to have resolved the aforemented [cmake error](https://github.com/nasa/fprime/issues/1163#issuecomment-999941989) without having to set the `RPI_TOOLCHAIN_DIR` environment variable. https://github.com/capsulecorplab/fprime/commit/51f50064588159345cfc26838b108d789076bb9e

astroesteban, 2022-05-08:
Be aware that my Raspberry Pi toolchain is for the [Raspberry Pi OS](https://www.raspberrypi.com/software/).

capsulecorplab, 2022-05-11:
Desperately attempting to get @astroesteban's [raspberrypi.cmake](https://github.com/nasa/fprime/issues/1163#issuecomment-1007810914) system to work, but ran into #1446

capsulecorplab, 2022-05-16:
@LeStarch Managed to successfully use @astroesteban's [raspberrypi cmake build system](https://github.com/nasa/fprime/issues/1163#issuecomment-1007810914) to build and run both the blinking LED example and Math components for performing math operations on my Raspberry Pi 3 running raspbian lite Jessie (legacy).

See https://github.com/capsulecorplab/fprime/tree/cross-compile-ref-with-mathcomponent-for-rpi-using-%40astroesteban-cmake-v3-0-0

Still working on the PR to update the CI, but I'm thinking we should proceed with using @astroesteban's cmake build system for the raspberry pi, as it doesn't require having to set the `RPI_TOOLCHAIN_DIR` environment variable. Thoughts?

LeStarch, 2022-05-17:
This toolchain requires that the RPI tools be on the path and uses whatever it finds.  If we go with this route, we'll need to update the the RPI README to change how the tools are installed.  @timcanham do you have an opinion on this?

timcanham, 2022-05-17:
Can we anchor the path with an externally settable variable? This approach makes it more difficult to specify which version of the tools works. Something like:

```
####
# Raspberry PI Toolchain
#
# This is a toolchain for the Raspberry Pi. This toolchain can be used ot build
# against the Raspberry Pi embedded Linux target. In order to use this toolchain,
# the Raspbery Pi cross-compiler should be installed on a Linux host. These
# tools are installable as follows:
#   sudo apt install gcc-arm-linux-gnueabihf g++-arm-linux-gnueabihf  gdb-multiarch
####

# Set the system information
set(CMAKE_SYSTEM_NAME       Linux) 
set(CMAKE_SYSTEM_PROCESSOR  arm)
set(CMAKE_SYSTEM_VERSION    1)

set (RPI_TOOL_BASE $ENV{RPI_TOOL_DIR}) # Ex. Set to /opt/rpi/tools in user environment or settings.ini
# Set the GNU ARM toolchain
set(CMAKE_AR            ${RPI_TOOL_BASE }/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/arm-linux-gnueabihf-ar)
set(CMAKE_C_COMPILER    ${RPI_TOOL_BASE }/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/arm-linux-gnueabihf-gcc)
set(CMAKE_CXX_COMPILER  ${RPI_TOOL_BASE }/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/arm-linux-gnueabihf-g++)
set(CMAKE_ASM_COMPILER  ${RPI_TOOL_BASE }/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/arm-linux-gnueabihf-as)
set(CMAKE_OBJCOPY       ${RPI_TOOL_BASE }/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/arm-linux-gnueabihf-objcopy)
set(CMAKE_OBJDUMP       ${RPI_TOOL_BASE }/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/arm-linux-gnueabihf-objdump)
set(CMAKE_C_GDB         ${RPI_TOOL_BASE }/arm-bcm2708/arm-rpi-4.9.3-linux-gnueabihf/gdb-multiarch)

# Configure the find commands for finding the toolchain
set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM   NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY   NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE   NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE   NEVER)
```

(I don't know if the file is actually correct, just an example)

LeStarch, 2022-05-20:
This would bring us full circle back to tHe original PI tool chain....Which doesn't support the on path programs

LeStarch, 2022-05-20:
Here is a trick I learned from looking into arm64 (aarm64).

```
find_program(CMAKE_C_COMPILER NAMES aarch64-linux-gnu-gcc aarch64-none-linux-gnu-gcc PATHS ENV AARCH_DIR PATH_SUFFIXES bin REQUIRED)
find_program(CMAKE_CXX_COMPILER NAMES aarch64-linux-gnu-g++ aarch64-none-linux-gnu-g++ PATHS ENV AARCH_DIR PATH_SUFFIXES bin REQUIRED)
message(STATUS "[aarch64-linux] C Compiler: ${CMAKE_C_COMPILER}")
message(STATUS "[aarch64-linux] CXX Compiler: ${CMAKE_CXX_COMPILER}")
```

This will search for the tool, of the given name.  First it searches at the environment variable (AARCH_DIR) then, if not set or not found, it will use the PATH to find the tool.

This style toolchain will work for:
1. OS package tools
2. tools added specifically to the PATH
3. Custom tools at a location driven by the operating system.

There are multiple NAMES because the aarch64 binarieschanges between versions.

capsulecorplab, 2022-06-03:
Any particular reason we'd want the toolchain to compile for 64-bit ARM as opposed to 32-bit? I was under the impression it should use the 32-bit, since that's what the (deprecated) rpi toolchain was using.

capsulecorplab, 2022-06-03:
In either case (32-bit or 64-bit), any preference as to where we want to be downloading the ARM toolchain from? I'm thinking we should update the CI to download/extract/install the toolchain directly from the source tarball off the ARM developer website https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/downloads

LeStarch, 2022-06-03:
F´  doesn't.  I have been working on an arm64 toolchain on the side.

LeStarch, 2022-06-03:
I haven't had luck with the arm website, so I have used Linaro. However, if structured with the ability to set a base, the system packages, linaro, or arm would be picked-up.

capsulecorplab, 2022-06-03:
Just tried generating build cache for RPI with https://github.com/capsulecorplab/fprime/commit/635867ac88f281de59a8d4cc5b894234d32803fc, but ran into the `fpp-tools version incompatible` issue https://github.com/nasa/fprime/issues/1478

capsulecorplab, 2022-07-21:
RPI build works now after updating my development environment with pinned python packages using the `requirements.txt` from v3.1.0 release! Opened PR #1578 for review

LeStarch, 2022-09-26:
Fixed!  Nice work @capsulecorplab!

Not sure this panned out to be an "Easy First Issue" but you got it done!

PR #1578 (2022-09-19): Update cross compile toolchain for raspberrypi in cmake & RPI CI
| | |
|:---|:---|
|**_Originating Project/Creator_**| |
|**_Affected Component_**| RPI |
|**_Affected Architectures(s)_**| |
|**_Related Issue(s)_**| #1163  |
|**_Has Unit Tests (y/n)_**| |
|**_Builds Without Errors (y/n)_**| y |
|**_Unit Tests Pass (y/n)_**| |
|**_Documentation Included (y/n)_**|  |

---
## Change Description

- Updates cross compile toolchain for raspberry pi by first checking `RPI_TOOLCHAIN_DIR` environment variable, then by checking PATH, in cmake toolchain for raspberry pi.
- Updates CI build for RPI to use  [... cut]
closed 2022-09-26 by LeStarch