nasa/fprime #1165: Separate Modules and Ports/Types

LeStarch, 2021-12-17:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

Ports and types are defined within the same module as components.

ThibFrgsGmz, 2022-01-04:
Could you elaborate? Do you have more information on what you are expecting?

LeStarch, 2022-01-05:
@ThibFrgsGmz yes!  

We need to survey the FPP files we have, looking for cases where a single FPP file defines ports/types **and** a component in the same file.  The FileDownlink is the example that raised red-flags:

Ports/Types defined inside:

https://github.com/nasa/fprime/blob/0b669e108b43bc61c699acf4925ac783fbdc7f12/Svc/FileDownlink/FileDownlink.fpp#L12

https://github.com/nasa/fprime/blob/0b669e108b43bc61c699acf4925ac783fbdc7f12/Svc/FileDownlink/FileDownlink.fpp#L18

Component defined in the same file:
https://github.com/nasa/fprime/blob/0b669e108b43bc61c699acf4925ac783fbdc7f12/Svc/FileDownlink/FileDownlink.fpp#L32

LeStarch, 2022-01-05:
This is a problem because components should not have link-time dependencies on other components.  This is a selling point of F´.  However, if a module (e.g. Svc_FileDownlink) defines ports (the interface between components) in the file as a component then there will be a link dependency.

LeStarch, 2022-01-05:
Here are the steps to finish this ticket:

1. List occurrences where ports and components are defined in the same FPP file.
2. List occurrences where types and components are defined in the same file. Note: types for internal use in the component only are ok, hence listing this separately from number 1
3. Review lists and determine which files need to be broken up
4. Develop a standard pattern to apply as the fix
5. Split files from item 3 by pattern in 4

If you want to tackle this one, post the results from item 1 and item 2 here, and we can work through 3-4 together!

ThibFrgsGmz, 2022-01-05:
@LeStarch  Since I haven't had a chance to look at F Prime 3.0 yet, this could serve as a first introduction, especially since it deals with the new main component, FPP. (The recent installation issues I've seen in the tickets/discussions make me dread this moment 😅)

The idea of taking over the ticket with your help appeals to me, and it could probably help me work on adding RTEMS 5.0 support to the project. 😀

I could start looking this weekend, but since I'm still finishing a research paper for a few more weeks (only evenings and weekends), I can't promise that the problem will be solved quickly. 🙁

LeStarch, 2022-01-05:
@ThibFrgsGmz for now I am trying to focus on non-"Easy First Issue" tickets, and there are other items that are more urgent....so I don't know it needs to be done super quickly.

I'll do my best to help, but I request you keep notes here in this issue.  That way the project is aware of the current state of the issue.

ThibFrgsGmz, 2022-01-09:
List of all fpp files in the project :
```sh
fprime$ find -O3 -L . -name "*.fpp"
./cmake/test/data/test-fprime-library/TestLibrary/TestComponent/TestComponent.fpp
./cmake/test/data/test-fprime-library2/TestLibrary2/TestComponent/TestComponent.fpp
./config/AcConstants.fpp
./config/FpConfig.fpp
./config/locs.fpp
./docs/Tutorials/MathComponent/MathPorts/MathPorts.fpp
./docs/Tutorials/MathComponent/MathReceiver/MathReceiver.fpp
./docs/Tutorials/MathComponent/MathSender/MathSender.fpp
./docs/Tutorials/MathComponent/MathTypes/MathTypes.fpp
./docs/Tutorials/MathComponent/Top/instances.fpp
./docs/Tutorials/MathComponent/Top/topology.fpp
./Drv/BlockDriver/BlockDriver.fpp
./Drv/ByteStreamDriverModel/ByteStreamDriverModel.fpp
./Drv/DataTypes/DataTypes.fpp
./Drv/GpioDriverPorts/GpioDriverPorts.fpp
./Drv/I2cDriverPorts/I2cDriverPorts.fpp
./Drv/LinuxGpioDriver/LinuxGpioDriver.fpp
./Drv/LinuxI2cDriver/LinuxI2cDriver.fpp
./Drv/LinuxSerialDriver/LinuxSerialDriver.fpp
./Drv/LinuxSpiDriver/LinuxSpiDriver.fpp
./Drv/locs.fpp
./Drv/SerialDriverPorts/SerialDriverPorts.fpp
./Drv/SpiDriverPorts/SpiDriverPorts.fpp
./Fpp/locs.fpp
./Fpp/ToCpp.fpp
./Fw/Buffer/Buffer.fpp
./Fw/Cmd/Cmd.fpp
./Fw/Com/Com.fpp
./Fw/locs.fpp
./Fw/Log/Log.fpp
./Fw/Prm/Prm.fpp
./Fw/Time/Time.fpp
./Fw/Tlm/Tlm.fpp
./Fw/Types/Types.fpp
./Ref/locs.fpp
./Ref/PingReceiver/PingReceiver.fpp
./Ref/RecvBuffApp/RecvBuffApp.fpp
./Ref/SendBuffApp/SendBuffApp.fpp
./Ref/SignalGen/SignalGen.fpp
./Ref/Top/instances.fpp
./Ref/Top/topology.fpp
./RPI/RpiDemo/RpiDemo.fpp
./RPI/Top/instances.fpp
./RPI/Top/topology.fpp
./Svc/ActiveLogger/ActiveLogger.fpp
./Svc/ActiveRateGroup/ActiveRateGroup.fpp
./Svc/ActiveTextLogger/ActiveTextLogger.fpp
./Svc/AssertFatalAdapter/AssertFatalAdapter.fpp
./Svc/BufferLogger/BufferLogger.fpp
./Svc/BufferManager/BufferManager.fpp
./Svc/CmdDispatcher/CmdDispatcher.fpp
./Svc/CmdSequencer/CmdSequencer.fpp
./Svc/ComLogger/ComLogger.fpp
./Svc/ComSplitter/ComSplitter.fpp
./Svc/Cycle/Cycle.fpp
./Svc/Deframer/Deframer.fpp
./Svc/Fatal/Fatal.fpp
./Svc/FatalHandler/FatalHandler.fpp
./Svc/FileDownlink/FileDownlink.fpp
./Svc/FileManager/FileManager.fpp
./Svc/FileUplink/FileUplink.fpp
./Svc/Framer/Framer.fpp
./Svc/GenericHub/GenericHub.fpp
./Svc/GenericRepeater/GenericRepeater.fpp
./Svc/GroundInterface/GroundInterface.fpp
./Svc/Health/Health.fpp
./Svc/LinuxTimer/LinuxTimer.fpp
./Svc/locs.fpp
./Svc/PassiveConsoleTextLogger/PassiveConsoleTextLogger.fpp
./Svc/Ping/Ping.fpp
./Svc/PolyDb/PolyDb.fpp
./Svc/PolyIf/PolyIf.fpp
./Svc/PrmDb/PrmDb.fpp
./Svc/RateGroupDriver/RateGroupDriver.fpp
./Svc/Sched/Sched.fpp
./Svc/Seq/Seq.fpp
./Svc/StaticMemory/StaticMemory.fpp
./Svc/SystemResources/SystemResources.fpp
./Svc/Time/Time.fpp
./Svc/TlmChan/TlmChan.fpp
./Svc/TlmPacketizer/TlmPacketizer.fpp
./Svc/WatchDog/WatchDog.fpp
```

According to my analysis the components that raise the red flags are:

- [Drv/ByteStreamDriverModel/ByteStreamDriverModel.fpp](https://github.com/nasa/fprime/blob/devel/Drv/ByteStreamDriverModel/ByteStreamDriverModel.fpp)
- [Ref/RecvBuffApp/RecvBuffApp.fpp](https://github.com/nasa/fprime/blob/devel/Ref/RecvBuffApp/RecvBuffApp.fpp)
- [Ref/SignalGen/SignalGen.fpp](https://github.com/nasa/fprime/blob/devel/Ref/SignalGen/SignalGen.fpp)
- [Svc/FileDownlink/FileDownlink.fpp](https://github.com/nasa/fprime/blob/devel/Svc/FileDownlink/FileDownlink.fpp)
- [Svc/SystemResources/SystemResources.fpp](https://github.com/nasa/fprime/blob/devel/Svc/SystemResources/SystemResources.fpp)

---

Step 1: list of occurrences where ``ports`` and ``components`` are defined in the same ``FPP`` file:

- [Drv/ByteStreamDriverModel/ByteStreamDriverModel.fpp](https://github.com/nasa/fprime/blob/devel/Drv/ByteStreamDriverModel/ByteStreamDriverModel.fpp) : ``enum`` ``port`` ``component``
- [Svc/FileDownlink/FileDownlink.fpp](https://github.com/nasa/fprime/blob/devel/Svc/FileDownlink/FileDownlink.fpp) : ``enum`` ``struct`` ``port`` ``component``

Step 2: list of occurrences where ``types`` and ``components`` are defined in the same ``FPP`` file:

- [Ref/RecvBuffApp/RecvBuffApp.fpp](https://github.com/nasa/fprime/blob/devel/Ref/RecvBuffApp/RecvBuffApp.fpp) : ``enum`` ``struct`` ``component``
- [Ref/SignalGen/SignalGen.fpp](https://github.com/nasa/fprime/blob/devel/Ref/SignalGen/SignalGen.fpp) : ``struct`` ``array`` ``enum`` ``component``
- [Svc/SystemResources/SystemResources.fpp](https://github.com/nasa/fprime/blob/devel/Svc/SystemResources/SystemResources.fpp) : ``enum`` ``component``

LeStarch, 2022-03-22:
@ThibFrgsGmz I didn't notice this was updated. This is good work.  Here is my analysis:

Must split:
[Svc/FileDownlink/FileDownlink.fpp](https://github.com/nasa/fprime/blob/devel/Svc/FileDownlink/FileDownlink.fpp) : enum struct port component

Probably OK:
1. [Drv/ByteStreamDriverModel/ByteStreamDriverModel.fpp](https://github.com/nasa/fprime/blob/devel/Drv/ByteStreamDriverModel/ByteStreamDriverModel.fpp) : enum port component

This package is only a "model" (no code) thus there will be no CPP dependency between packages using this for the same reason that ports (also model only) don't cause this problem.

2. [Svc/SystemResources/SystemResources.fpp](https://github.com/nasa/fprime/blob/devel/Svc/SystemResources/SystemResources.fpp) 
3. [Ref/SignalGen/SignalGen.fpp](https://github.com/nasa/fprime/blob/devel/Ref/SignalGen/SignalGen.fpp)
4. [Ref/RecvBuffApp/RecvBuffApp.fpp](https://github.com/nasa/fprime/blob/devel/Ref/RecvBuffApp/RecvBuffApp.fpp)
These types are only used internally for the command/telemetry/events definitions not "shared" with other models. No need to separate.

LeStarch, 2022-04-12:
@ThibFrgsGmz I am going to take this issue back from you and correct the problem as state here. Is this OK?

LeStarch, 2022-04-13:
I closed this out.  See PR.

ThibFrgsGmz, 2022-04-13:
OK, sorry, I completely lost track of the ticket!

PR #1390 (2022-04-13): lestarch: splitting FileDownlink from types. Fixes: #1165
| | |
|:---|:---|
|**_Originating Project/Creator_**| |
|**_Affected Component_**|  |
|**_Affected Architectures(s)_**|  |
|**_Related Issue(s)_**|  |
|**_Has Unit Tests (y/n)_**|  |
|**_Builds Without Errors (y/n)_**|  |
|**_Unit Tests Pass (y/n)_**|  |
|**_Documentation Included (y/n)_**|  |

---
## Change Description

Fixes #1165.
commit 191aa89d (2022-04-13): lestarch: splitting FileDownlink from types. Fixes: #1165 (#1390)
closed 2022-04-13 by LeStarch