nasa/fprime #1170: Effeciency Improvements in FPP/CMake Integration

LeStarch, 2021-12-20:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Further improve the speed of FPP and CMake integration.

## Rationale

Try to cut down overhead of FPP.

litghost, 2021-12-22:
Hi, I just saw the release notes for [v2.1.0](https://github.com/nasa/fprime/releases/tag/v2.1.0) and saw that the documentation is reflecting an integration of fpp into fprime in v3.0.0.  I was excited to try out the new fpp syntax, and was following the new Math component demo.  Something I quickly noticed is that CMake configure and build times from v2.1.0 to v3.0.0 greatly increased.  Specifically a ~25x increase in configure time (from ~2.4 seconds to ~61 seconds) and a ~3x increase in build time (from 5 seconds to 14 seconds with `-j16`).

Worse still, adding the `MathTypes` component added ~0.5-1 seconds to the configure time, leading me to strongly suspect that there is high linear cost to the number of components present in the current application.  A quick experiment where I created and added 10 simple components added ~8 seconds to the configure time, confirming the initial suspicion.

This appears to be a known issue (hence this issue), but is there a time frame for addressing it?  Waiting ~60 seconds to get a failure message due to a top level wiring problem is a frustrating development experience.

System configuration:
OS: Linux (Ubuntu 20.04)
```
uname -a
Linux kmachine 5.11.0-41-generic #45~20.04.1-Ubuntu SMP Wed Nov 10 10:20:10 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

Java:
```
java --version
openjdk 17.0.1 2021-10-19
OpenJDK Runtime Environment (build 17.0.1+12-Ubuntu-120.04)
OpenJDK 64-Bit Server VM (build 17.0.1+12-Ubuntu-120.04, mixed mode, sharing)
```

CMake:
```
cmake --version
cmake version 3.16.3

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

fprime-util:
```
Installing collected packages: fprime-tools
Successfully installed fprime-tools-3.0.0
```

LeStarch, 2021-12-22:
Did you see an initial speed-up after the first generate stage?  It should be faster after the initial generate.

litghost, 2021-12-22:
It's still very expensive.  I'm currently seeing ~15-20 seconds after the initial generate.  Both of those numbers are above comfort level.  And when attempting to port a pre-3.0.0 release, I'm stuck in initial generation so I have to repeat those 60 seconds.

litghost, 2021-12-22:
For example:
```
touch Top/topology.fpp 
/usr/bin/time -v fprime-util build -j16
```
Takes:
```
	Elapsed (wall clock) time (h:mm:ss or m:ss): 0:16.87
```

For context in v2.1.0, running:
```
touch Top/RefTopologyAppAi.xml 
/usr/bin/time -v fprime-util build -j16
```
takes:
```
	Elapsed (wall clock) time (h:mm:ss or m:ss): 0:05.84
```

LeStarch, 2021-12-22:
Yes, I agree that it is quite expensive to run these tools.  I am investigating solutions.

LeStarch, 2021-12-22:
@litghost what OS or OSes do you use?  Looking into various solutions, which may be OS dependent.

litghost, 2022-01-04:
Linux primarily

LeStarch, 2022-01-06:
Circling back here, I was able to reduce the time required to run FPP by 50%. I am currently focusing on `generate` as it is the linear step.  `-j` can engage parallelism to reduce the pain of the build step.

The needed F´ improvements are here: https://github.com/LeStarch/fprime/tree/update/speed-machine

The needed FPP tools are released as a package here: https://github.com/LeStarch/fpp/releases/tag/v0.4.0-experimental

Simply add those tools to your path, and ensure one is using the above F´ branch to successfully engage these tools and these faster pre-compiled binaries should run.  As a fun note, Java should not be required for these tools.

LeStarch, 2022-01-06:
This F´ must do to help solve this problem:

1. Publish and use these faster tool equivalents
2. Allow users to decline the "automatic install" and defer to tools installed path
3. Focus on backwards compatibility such that tools version check can be "at least version XYZ" instead of "exactly XYZ"

To fully solve the problem, additional work would be needed to improve the FPP implementation's runtime efficiency as the binary compiled code is still ~4x slower than the XML equivalents.

litghost, 2022-01-06:
> Allow users to decline the "automatic install" and defer to tools installed path

For reference, I was ensuring that my test cases were not in the "automatic install" path.  It would be good to be able to pass an argument to Fprime initialization to **never** perform automatic installation.  Something akin to:
 ```
find_package(FPP REQUIRED)
find_package(Fprime REQUIRED USE_INSTELLED_FPP)
```

LeStarch, 2022-01-24:
As an update, we have been looking into ways to resolve the overall efficiency issues with `v3.0.0` and should see improvement in future release. I'll post back here as information becomes available.

LeStarch, 2022-02-04:
Here is a tracking issue for a larger cmake work that will hopefully help on the performance front: #1260.  It has been a lot of work to chart a course through cmake + autocode that works correctly, complies with best practices, and is faster.

LeStarch, 2022-03-22:
Closing this ticket at it is tracked in #1260 and PRs submitted.