nasa/fprime #1182: Revive Small Platform Support

LeStarch, 2021-12-21:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

AVR and small ARM compilations are failing since v2.1.  We should revive support.

astroesteban, 2022-02-10:
@LeStarch Would you consider the fprime-arduino repo a good overall reference for porting F Prime `v3.0.0` to a new bare metal embedded microcontroller? Or is there another project in the fprime-community that you recommend?

LeStarch, 2022-02-10:
@astroesteban fprime-arduino is the repository that is failing.  As far as I am aware there are no current small baremetal systems using F´. The fprime-arduino toolchain approach should work, it just needs an fpp-ification of its components.

@SterlingPeet was looking into refactoring fprime-arudino at some point, so we should see if changes are in progress there.

astroesteban, 2022-02-11:
I see. Good to know. I'll spend some time over the next couple of months porting F' `v3.0.0` to the Raspberry Pi Pico. I'll make sure to document any blockers or quirks I come across.

not7cd, 2022-05-20:
What is the current status of this issue? I'm interested in running f-prime on pi pico. Is it something worth pursing alone?

LeStarch, 2022-09-29:
Note: @SterlingPeet and I are working on upgrading `fprime-arduino` to generally support arduino-framework boards. However, along with the other issue....this is being done in `fprime-community` and as such this issue should be closed.

closed 2022-09-29 by LeStarch