nasa/fprime #1190: README.md cluttered with release notes

ThibFrgsGmz, 2021-12-27:
| | |
|:---|:---|
|**_F´ Version_**| devel |
|**_Affected Component_**|  README.md |
---
## Problem Description

The release notes are starting to clutter the project's README.md file.

## How to Reproduce

Successfully navigate to the bottom of the release notes section of the README.md file. 😀

## Expected Behavior

I think of several ways to make it more digestible. We should empty the section by exporting the information elsewhere and redirecting the reader to that new location. The section would only include a link to the change log and a high-level summary, which would only be updated when there are major changes.

For the centralized change history, I think we should redirect to https://github.com/nasa/fprime/releases. Another option is to copy/paste the section into a new file called CHANGELOG.md in the root directory of the project.

## Miscellaneous

I also recommend renaming the "F´ Release Notes" to simply "Release Notes" because including the project name here is unnecessary.

ThibFrgsGmz, 2021-12-28:
When I compare it to other popular Open Source repositories, I notice that they don't have a "Release Notes" section in their README.

- React from Facebook https://github.com/facebook/react
- Rust language https://github.com/rust-lang/rust
- Vuejs https://github.com/vuejs/vue

They let the user to go directly to the "Releases" section of the GitHub project to obtain the necessary information.

So I'm wondering if we should just leave the Release Notes section out of the README.

LeStarch, 2022-01-03:
I think we'd prefer to use the `CHANGELOG.md` file and for now, we can track that data there.   Release notes are quite nice, but it is also nice to have this information in the repository directly so moving the items out to a new file would achieve this balance and clean up the README.md.

I am going to mark this as an "Easy First Issue" and "Documentation".

commit 65d960f1 (2022-01-05): Fix #1190 README.md cluttered with release notes (#1200)

* Exporting the update note to a specific file

* Renaming Release Notes section

* Chronological reorganization of the release notes

* Add redirection to CHANGELOG.md
commit f4ff5c08 (2022-06-29): Fix #1190 README.md cluttered with release notes (#1200)

* Exporting the update note to a specific file

* Renaming Release Notes section

* Chronological reorganization of the release notes

* Add redirection to CHANGELOG.md
closed 2022-01-05 by LeStarch