nasa/fprime #1193: Having trouble generating fprime-util when building the Ref application

arshiuh, 2022-01-03:
Hello! I am trying to follow the Getting Started section and checking my F' Installation, and it seems that the autocoder did not install properly. When I cd into the Ref folder and enter `fprime-util generate`. It says `fprime-util: command not found`. I am most likely installed it incorrectly, but I am slightly stuck on where to go from here.

**System Versions:**
Running Linux Ubuntu 20.04.3 on VirtualBox

Java Version
`openjdk 11.0.13 2021-10-19
OpenJDK Runtime Environment (build 11.0.13+8-Ubuntu-0ubuntu1.20.04)
OpenJDK 64-Bit Server VM (build 11.0.13+8-Ubuntu-0ubuntu1.20.04, mixed mode, sharing)`

CMake
`cmake version 3.16.3
CMake suite maintained and supported by Kitware (kitware.com/cmake).
`

Bash
`GNU bash, version 4.4.0(1)-release (x86_64-unknown-linux-gnu)
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>`

CLang
`clang version 10.0.0-4ubuntu1 
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin`

Python & PIP
`Python 3.8.10
pip 20.0.2 from /usr/lib/python3/dist-packages/pip (python 3.8)
`

ThibFrgsGmz, 2022-01-03:
Do you have the python virtual environment activated?

LeStarch, 2022-01-03:
`fprime-util: command not found` is almost always caused by one of two issues:

1. `fprime-util` was installed, but not in path.  On some systems, `pip` places user installed tools in `.local` and this needs to be added to the path.
2. `pip install fprime-tools` was not run.

Point 1 above is not noted in the installation guide.  We'll need to fix that.

arshiuh, 2022-01-04:
@LeStarch Point 1 seems to have fixed it. Thank you very much!

LeStarch, 2022-01-04:
We need to add a note in install instructions to add .local to path or run as an admin.

closed 2022-03-22 by LeStarch