nasa/fprime #1195: No rule to make target 'Ref_SignalGen_testimpl' with fprime-util impl --ut in tutorial

dimethylzinc, 2022-01-03:
| | |
|:---|:---|
|**_F´ Version_**| 85fa25b |
|**_Affected Component_**| fprime-util, tutorial  |
---
## Problem Description
`fprime-util impl --ut` fails on vanilla tutorial installation:
```
(fprime-venv) user@computer /localdata2/user/fprime1/fprime/Ref/SignalGen % fprime-util impl --ut
[WARNING] Failed to find settings file: /localdata2/user/fprime1/fprime/Ref/settings.ini
make: *** No rule to make target 'Ref_SignalGen_testimpl'.  Stop.
[CMAKE] CMake failed to detect target, attempting CMake cache refresh and retry
make: *** No rule to make target 'Ref_SignalGen_testimpl'.  Stop.
[ERROR] CMake erred with return code 2
```

## How to Reproduce

1. Fresh install as per INSTALL.md (functioning HTML GUI)
2. Getting started tutorial as per Tutorial.md, skipping raspberry pi build
```
cd fprime/Ref
fprime-util generate
cd fprime/Ref/SignalGen
fprime-util impl
```
(Runs successfully to 100%, generates -template files)
3. Attempt to generate unit test implementation stubs
```
fprime-util impl --ut
```

## Expected Behavior

Stubbed implementation files for the unit tests are created.

LeStarch, 2022-01-04:
This is currently broken.  A work around is to do the following:

```
fprime-util info --ut
cd <build cache>
make <target>_testimpl
```

LeStarch, 2022-01-04:
Note: should also add this to UTs for fprime-util as this should have been caught automatically.

LeStarch, 2022-01-04:
Also missing TestMain.cpp?

closed 2022-01-05 by kevin-f-ortega

Emil808, 2022-01-19:
I am encountering this same problem today

LeStarch, 2022-01-19:
@Emil808 we released the new version of tools that would fix this yesterday.  Are you running fprime-tools v3.0.1?

Emil808, 2022-01-19:
@LeStarch I did pip install upgrade for fprime-tools and is now up to date with v3.0. 

On a fresh clone of fprime, I am able to generate the unit test. On the fork that I'm working, with pulling the latest update and merging into a wip branch, I still cannot generate the uts.

capsulecorplab, 2022-05-10:
> This is currently broken. A work around is to do the following:
> 
> ```
> fprime-util info --ut
> cd <build cache>
> make <target>_testimpl
> ```

I'm trying to do this for the raspberrypi, but `make raspberrypi_testimpl` returns a `make: *** No rule to make target 'linux_testimpl'.  Stop.` message. I generated the unit tests in my linux environment and am attempting to build on my raspberrypi