nasa/fprime #1201: cookiecutter exceptions RepositoryNotFound

kevin-f-ortega, 2022-01-04:
| | |
|:---|:---|
|**_F´ Version_**| 0b669e108b43bc61c699acf4925ac783fbdc7f12 |
|**_Affected Component_**|  N/A |
---
## Problem Description

Cannot create a new component with the fprime utility.

## How to Reproduce

1. fprime-util generate
2. fprime-util new --component native

## Expected Behavior

Expected to be prompted for information regarding new component. Instead, I got these errors

```
fprime-util new --component native
[WARNING] Failed to find settings file: /home/ortega/abc/fprime/settings.ini
[WARNING] **** fprime-util new is prototype functionality ****
[INFO] Cookiecutter source: using builtin

Traceback (most recent call last):
  File "/home/ortega/env/abc-venv/bin/fprime-util", line 8, in <module>
    sys.exit(main())
  File "/home/ortega/env/abc-venv/lib/python3.8/site-packages/fprime/util/__main__.py", line 14, in main
    return fprime.util.build_helper.utility_entry(args=sys.argv[1:])
  File "/home/ortega/env/abc-venv/lib/python3.8/site-packages/fprime/util/build_helper.py", line 155, in utility_entry
    runners[parsed.command](build, parsed, cmake_args, make_args)
  File "/home/ortega/env/abc-venv/lib/python3.8/site-packages/fprime/util/cli.py", line 112, in template
    return new_component(build.deployment, parsed.platform, parsed.verbose, build)
  File "/home/ortega/env/abc-venv/lib/python3.8/site-packages/fprime/fbuild/interaction.py", line 265, in new_component
    cookiecutter(source, extra_context={"component_namespace": deployment.name})
  File "/home/ortega/env/abc-venv/lib/python3.8/site-packages/cookiecutter/main.py", line 67, in cookiecutter
    repo_dir, cleanup = determine_repo_dir(
  File "/home/ortega/env/abc-venv/lib/python3.8/site-packages/cookiecutter/repository.py", line 131, in determine_repo_dir
    raise RepositoryNotFound(
cookiecutter.exceptions.RepositoryNotFound: A valid repository for "/home/ortega/env/abc-venv/lib/python3.8/site-packages/fprime/fbuild/../cookiecutter_templates/cookiecutter-fprime-component" could not be found in the following locations:
/home/ortega/env/abc-venv/lib/python3.8/site-packages/fprime/fbuild/../cookiecutter_templates/cookiecutter-fprime-component
/home/ortega/env/abc-venv/lib/python3.8/site-packages/fprime/fbuild/../cookiecutter_templates/cookiecutter-fprime-component
```

LeStarch, 2022-01-05:
Also need to update the utility to use FPP.

LeStarch, 2022-01-05:
Closing in favor of: https://github.com/fprime-community/fprime-tools/issues/34

closed 2022-01-05 by LeStarch