nasa/fprime #1203: rpiDemo.RD_SetLed command not working

capsulecorplab, 2022-01-05:
| | |
|:---|:---|
|**_F´ Version_**| v3.0.0 |
|**_Affected Component_**| RPI demo |
---
## Problem Description

`rpiDemo.RD_SetLed` command mnemonic with value `BLINKING` or `OFF` yields `[ERROR] Deframing checksum validation failed` on Raspberry Pi 3B v1.2. LED on GPIO21 continuously blinks, regardless of command sent to raspberry pi.

```
pi@raspberrypi:~ $ ./RPI -a 192.168.0.76 -p 50000
Hit Ctrl-C to quit
EVENT: (500) (2:1641339260,548779) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x1f4 registered to port 0 slot 0
EVENT: (500) (2:1641339260,548864) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x1f5 registered to port 0 slot 1
EVENT: (500) (2:1641339260,548903) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x1f6 registered to port 0 slot 2
EVENT: (500) (2:1641339260,548940) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x1f7 registered to port 0 slot 3
EVENT: (500) (2:1641339260,548977) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2bc registered to port 1 slot 4
EVENT: (500) (2:1641339260,549014) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2bd registered to port 1 slot 5
EVENT: (500) (2:1641339260,549050) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2be registered to port 1 slot 6
EVENT: (500) (2:1641339260,549086) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2bf registered to port 1 slot 7
EVENT: (500) (2:1641339260,549122) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2c0 registered to port 1 slot 8
EVENT: (500) (2:1641339260,549158) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2c1 registered to port 1 slot 9
EVENT: (500) (2:1641339260,549194) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2c2 registered to port 1 slot 10
EVENT: (500) (2:1641339260,549230) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x2c3 registered to port 1 slot 11
EVENT: (500) (2:1641339260,549266) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x578 registered to port 2 slot 12
EVENT: (500) (2:1641339260,549303) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x57a registered to port 2 slot 13
EVENT: (500) (2:1641339260,549339) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x57b registered to port 2 slot 14
EVENT: (500) (2:1641339260,549375) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x708 registered to port 3 slot 15
EVENT: (500) (2:1641339260,549411) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x709 registered to port 3 slot 16
EVENT: (500) (2:1641339260,549447) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x70a registered to port 3 slot 17
EVENT: (500) (2:1641339260,549483) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x44c registered to port 4 slot 18
EVENT: (500) (2:1641339260,549518) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x44d registered to port 4 slot 19
EVENT: (500) (2:1641339260,549554) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x44e registered to port 4 slot 20
EVENT: (500) (2:1641339260,549591) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0x258 registered to port 5 slot 21
EVENT: (500) (2:1641339260,549627) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0xa8c registered to port 6 slot 22
EVENT: (500) (2:1641339260,549663) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0xa8d registered to port 6 slot 23
EVENT: (500) (2:1641339260,549699) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0xa8e registered to port 6 slot 24
EVENT: (500) (2:1641339260,549735) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0xa8f registered to port 6 slot 25
EVENT: (500) (2:1641339260,549771) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0xa90 registered to port 6 slot 26
EVENT: (500) (2:1641339260,549807) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0xa91 registered to port 6 slot 27
EVENT: (500) (2:1641339260,549843) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0xa96 registered to port 6 slot 28
EVENT: (500) (2:1641339260,549878) DIAGNOSTIC: (cmdDisp) OpCodeRegistered : Opcode 0xa97 registered to port 6 slot 29
EVENT: (606) (2:1641339260,549933) WARNING_HI: (prmDb) PrmFileReadError : Parameter file read failed in stage OPEN with record 0 and error 1
EVENT: (600) (2:1641339260,549993) WARNING_LO: (prmDb) PrmIdNotFound : Parameter ID 0xa8c not found
pthread_create: Insufficient permissions to create thread. May not set thread priority without permission. Operation not permitted
[WARNING] Insufficient Permissions:
[WARNING] Insufficient permissions to set task priority or set task CPU affinity on task TP_chanTlm. Creating task without priority nor affinity.
[WARNING] Please use no-argument <component>.start() calls, set priority/affinity to TASK_DEFAULT or ensure user has correct permissions for operating system.
[WARNING]      Note: future releases of fprime will fail when setting priority/affinity without sufficient permissions 

[WARNING] Task priority set and permissions unavailable. Discarding priority.
pthread_create: Insufficient permissions to create thread. May not set thread priority without permission. Operation not permitted
[WARNING] Insufficient Permissions:
[WARNING] Insufficient permissions to set task priority or set task CPU affinity on task TP_cmdDisp. Creating task without priority nor affinity.
[WARNING] Please use no-argument <component>.start() calls, set priority/affinity to TASK_DEFAULT or ensure user has correct permissions for operating system.
[WARNING]      Note: future releases of fprime will fail when setting priority/affinity without sufficient permissions 

[WARNING] Task priority set and permissions unavailable. Discarding priority.
pthread_create: Insufficient permissions to create thread. May not set thread priority without permission. Operation not permitted
[WARNING] Insufficient Permissions:
[WARNING] Insufficient permissions to set task priority or set task CPU affinity on task TP_cmdSeq. Creating task without priority nor affinity.
[WARNING] Please use no-argument <component>.start() calls, set priority/affinity to TASK_DEFAULT or ensure user has correct permissions for operating system.
[WARNING]      Note: future releases of fprime will fail when setting priority/affinity without sufficient permissions 

[WARNING] Task priority set and permissions unavailable. Discarding priority.
[WARNING] High task priority of 16384 being clamped to 99
pthread_create: Insufficient permissions to create thread. May not set thread priority without permission. Operation not permitted
[WARNING] Insufficient Permissions:
[WARNING] Insufficient permissions to set task priority or set task CPU affinity on task TP_ReceiveTask. Creating task without priority nor affinity.
[WARNING] Please use no-argument <component>.start() calls, set priority/affinity to TASK_DEFAULT or ensure user has correct permissions for operating system.
[... 98 lines cut ...]
[ERROR] Failed to send framed data: 1
[ERROR] Failed to send framed data: 1
EVENT: (200) (2:1641339012,559116) DIAGNOSTIC: (rateGroup10HzComp) RateGroupStarted : Rate group started.
EVENT: (300) (2:1641339012,559233) DIAGNOSTIC: (rateGroup1HzComp) RateGroupStarted : Rate group started.
EVENT: (2707) (2:1641339012,559359) ACTIVITY_HI: (rpiDemo) RD_LedBlinkState : LED blink state set to BLINKING
[ERROR] Failed to send framed data: 1
Connected to 192.168.0.76:50000 as a tcp client
[ERROR] Deframing checksum validation failed
```

## How to Reproduce

1. Flash [Raspberry Pi OS (Buster) 2021-12-02 Release](https://downloads.raspberrypi.org/raspios_oldstable_armhf/images/raspios_oldstable_armhf-2021-12-02/2021-12-02-raspios-buster-armhf.zip) using [dd](https://www.raspberrypi.com/documentation/computers/getting-started.html#installing-images-on-linux) to sd card.
2. Boot up and configure raspberry pi to connect to home WiFi network.
3. Enable SPI and serial interface via raspberry configuration menu, as per [board preparations](https://github.com/nasa/fprime/tree/devel/RPI#board-preparation). Disable serial console on raspberry pi, as per [disabling the linux serial console](https://www.raspberrypi.com/documentation/computers/configuration.html#disabling-the-linux-serial-console).
4. Connect LED to GPIO21.
3. Build & deploy `RPI` binary to raspberry pi, as per [build the software](https://github.com/nasa/fprime/tree/devel/RPI#build-the-software).
4. Run `fprime-gds -n --dictionary ./build-artifacts/raspberrypi/dict/RPITopologyAppDictionary.xml` on Ubuntu desktop.
5. Run `./RPI -a <ground computer's IP> -p 50000` on raspberry pi connected to same network as Ubuntu desktop.
6. Wait for tcp client to connect to server.
6. Send command `rpiDemo.RD_SetLed` with value `"BLINKING"` via ground station running on `localhost:5000`.

## Expected Behavior

Sending command mnemonic, `rpiDemo.RD_SetLed`, with value `OFF`, should turn off LED on GPIO21

LeStarch, 2022-01-05:
```
[ERROR] Deframing checksum validation failed
```

Is usually caused by a miss-match between an older `fprime-gds` and newer fprime code.  What is your fprime-gds version?

```
pip freeze | grep fprime-gds
```

capsulecorplab, 2022-01-05:
> ```
> [ERROR] Deframing checksum validation failed
> ```
> 
> Is usually caused by a miss-match between an older `fprime-gds` and newer fprime code. What is your fprime-gds version?
> 
> ```
> pip freeze | grep fprime-gds
> ```
Ah! `fprime-gds` was pointing to an older version, `fprime-gds-1.5.0`, outside my virtual environment. Running a `pip uninstall fprime-gds` outside my virtual environment and installing with `pip install fprime-gds` within my virtual environment seems to have fixed the issue. Thanks @LeStarch !

closed 2022-01-05 by capsulecorplab